lineage-rpi/android_kernel_brcm_rpi #5: Great work!Thanks!

andersling, 2020-08-16:
Hi,sir! Thanks for your big efffort !
 and btw,how to build it for  raspberry pi 4? would you do me a fovar to give some instructions? thanks!

KonstaT, 2020-08-16:
Standalone method described in this post (https://github.com/lineage-rpi/android_local_manifest/issues/8#issuecomment-625031841) works for building the kernel. I'll later push skeleton device trees for lineage-17.1 branch as well so kernel can be compiled as part of Android build too.

closed 2020-08-16 by KonstaT

andersling, 2020-08-16:
> Standalone method described in this post ([lineage-rpi/android_local_manifest#8 (comment)](https://github.com/lineage-rpi/android_local_manifest/issues/8#issuecomment-625031841)) works for building the kernel. I'll later push skeleton device trees for lineage-17.1 branch as well so kernel can be compiled as part of Android build too.

That is great!!! Big thanks to you ,sir!!!

andersling, 2020-09-03:
Hi,KonstaK sir! I found another exciting isue today:
  I build the 17.1 kernel (only the zImage),and replace the one in the boot partiction of  your 0813's ROOM. and then I can boot into the OS (event with the wifi on in the config.txt), and I can connect to some AP by manual! and the blooth is also working by connecting to some blue device. and the problem is  still exists: the wifi is not so stable! when the WiFi is ON, the desktop of lineageOS  frequently turn to black(empty screen,for about 1-2 seconds,and back to normal!) But, It doesen't reboot, it just switch on an off.
    ----------
   [  551.588964] type=1400 audit(1597321036.709:204): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=9704 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1 app=com.android.providers.blockednumber
[  551.611852] type=1400 audit(1597321036.709:205): avc: denied { open } for comm="main" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=9704 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1 app=com.android.providers.blockednumber
[  574.639670] binder: undelivered TRANSACTION_COMPLETE
[  574.644860] binder: undelivered transaction 467757, process died.
[ 2688.285842] init: Received control message 'interface_start' for 'android.hardware.wifi.supplicant@1.0::ISupplicant/default' from pid: 151 (/system/bin/hwservicemanager)
[ 2688.302462] init: starting service 'wpa_supplicant'...
[ 2688.316597] init: Created socket '/dev/socket/wpa_wlan0', mode 660, user 1010, group 1010
[ 2688.501891] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 2688.538032] init: Received control message 'interface_start' for 'android.hardware.wifi.offload@1.0::IOffload/default' from pid: 151 (/system/bin/hwservicemanager)
[ 2688.555379] init: Could not find 'android.hardware.wifi.offload@1.0::IOffload/default' for ctl.interface_start
[ 2719.283467] type=1400 audit(1597321036.709:205): avc: denied { open } for comm="main" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=9704 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1 app=com.android.providers.blockednumber
[ 2719.310023] type=1400 audit(1597323204.469:206): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=9720 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[ 2719.343858] type=1400 audit(1597323204.469:206): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=9720 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[ 2719.362499] type=1400 audit(1597323204.469:207): avc: denied { open } for comm="main" path="/dev/stune/top-app/tasks" dev="tmpfs" ino=9720 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[ 2719.611383] rpi_firmware_get_throttled: 1 callbacks suppressed
[ 2719.611393] Under-voltage detected! (0x00050005)
[ 2720.223877] type=1400 audit(1597323204.469:207): avc: denied { open } for comm="main" path="/dev/stune/top-app/tasks" dev="tmpfs" ino=9720 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[ 2720.243826] type=1400 audit(1597323205.409:208): avc: denied { search } for comm="RenderThread" name="dri" dev="tmpfs" ino=9383 scontext=u:r:system_app:s0 tcontext=u:object_r:gpu_device:s0 tclass=dir permissive=1
[ 2720.263356] type=1400 audit(1597323205.409:208): avc: denied { search } for comm="RenderThread" name="dri" dev="tmpfs" ino=9383 scontext=u:r:system_app:s0 tcontext=u:object_r:gpu_device:s0 tclass=dir permissive=1
[ 2720.282672] type=1400 audit(1597323205.409:209): avc: denied { read } for comm="RenderThread" name="uevent" dev="sysfs" ino=3750 scontext=u:r:system_app:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[ 2720.288965] init: Received control message 'interface_start' for 'android.hardware.graphics.mapper@3.0::IMapper/default' from pid: 151 (/system/bin/hwservicemanager)
[ 2720.302005] type=1400 audit(1597323205.409:209): avc: denied { read } for comm="RenderThread" name="uevent" dev="sysfs" ino=3750 scontext=u:r:system_app:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[ 2720.317267] init: Could not find 'android.hardware.graphics.mapper@3.0::IMapper/default' for ctl.interface_start
[ 2720.335631] type=1400 audit(1597323205.409:210): avc: denied { open } for comm="RenderThread" path="/sys/devices/platform/v3dbus/fec00000.v3d/uevent" dev="sysfs" ino=3750 scontext=u:r:system_app:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[ 2723.771320] rpi_firmware_get_throttled: 1 callbacks suppressed
[ 2723.771325] Voltage normalised (0x00000000)
[ 2726.823911] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[ 2727.375638] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 2727.933246] Under-voltage detected! (0x00050005)
[ 2730.011348] Voltage normalised (0x00000000)
[ 2742.493755] Under-voltage detected! (0x00050000)
[ 2744.571333] Voltage normalised (0x00000000)

 ------------------------------------- 

[   55.902834] type=1400 audit(1597320541.639:128): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=8548 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   55.920790] type=1400 audit(1597320541.639:129): avc: denied { open } for comm="main" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=8548 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1

[... cut ...]
[ 1088.022034] brcmfmac: brcmf_cfg80211_set_power_mgmt: Anders: ----------------------->power save disabled
[ 1088.560636] brcmfmac: brcmf_cfg80211_set_power_mgmt: Anders: ----------------------->power save enabled
[ 1088.811765] Under-voltage detected! (0x00050000)
[ 1088.826064] init: processing action (sys.sysctl.tcp_def_init_rwnd=*) from (/init.rc:823)
[ 1092.971357] Voltage normalised (0x00000000)
[ 1101.301687] Under-voltage detected! (0x00050005)
[ 1103.371370] Voltage normalised (0x00000000)
-------------------------------------

andersling, 2020-09-03:
And BTW, I can I build you 17.1 room? I follow the 15.1 tips, it builded ramdisk systemimg,verdor failed!
I can only build the kernel!
 Would you do me a favor to give some instruction, I hope that i can try some method to report something about this wifi to you ,so that you may refered it to resolve it.
thank you ,sir!
Hope you can pick time to reply me,thanks!

KonstaT, 2020-09-05:
Make sure you're using a proper power source (3A in case of Pi 4) because you're running into under-voltage.

There isn't much of a difference between the kernel in 0813 release and what is here currently (matches 0815 Android TV build at the moment). That shouldn't make any difference regarding wifi AFAICT.

There's source code and build instructions for my LineageOS 14.1 & 15.1 builds for Pi 3 (https://github.com/lineage-rpi/android_local_manifest/tree/cm-14.1 & https://github.com/lineage-rpi/android_local_manifest/tree/lineage-15.1, respectively). I haven't released the full device specific source code for my LineageOS 16.0 & 17.1 builds (both Pi 3 & Pi 4) because someone sold builds off of the sources I previously released without complying with open source licensing.