lineage-rpi/android_kernel_brcm_rpi #6: Can't boot up,the Debug msg ,thanks !

andersling, 2020-08-16:
Hi,sir! When I use you 20200813 rom for pi4,it keep rebooting and rebooting, i catch the debug messages from USB ,please take a look at it,thanks!

---------------------------------------------------------------
[   35.187724] type=1400 audit(1597320517.289:101): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   35.200637] init: Could not find 'vendor.lineage.livedisplay@2.0::IAutoContrast/default' for ctl.interface_start
[   35.219864] type=1400 audit(1597320517.289:101): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   35.230787] init: Received control message 'interface_start' for 'vendor.lineage.livedisplay@2.0::IDisplayColorCalibration/default' from pid: 146 (/system/bin/hwservicemanager)
[   35.247169] type=1400 audit(1597320517.289:102): avc: denied { open } for comm="main" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   35.263288] init: Could not find 'vendor.lineage.livedisplay@2.0::IDisplayColorCalibration/default' for ctl.interface_start
[   35.294351] init: Received control message 'interface_start' for 'vendor.lineage.livedisplay@2.0::IDisplayModes/default' from pid: 146 (/system/bin/hwservicemanager)
[   35.309852] init: Could not find 'vendor.lineage.livedisplay@2.0::IDisplayModes/default' for ctl.interface_start
[   35.320421] init: Received control message 'interface_start' for 'vendor.lineage.livedisplay@2.0::IReadingEnhancement/default' from pid: 146 (/system/bin/hwservicemanager)
[   36.851865] type=1400 audit(1597320517.289:102): avc: denied { open } for comm="main" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   36.873786] type=1400 audit(1597320518.979:103): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   36.894704] type=1400 audit(1597320518.979:103): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   36.915905] type=1400 audit(1597320518.979:104): avc: denied { open } for comm="main" path="/dev/stune/top-app/tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   37.175658] type=1400 audit(1597320518.979:104): avc: denied { open } for comm="main" path="/dev/stune/top-app/tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   37.198532] type=1400 audit(1597320519.309:105): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1 app=com.android.providers.media
[   37.670057] 8<--- cut here ---
[   37.674677] Unable to handle kernel NULL pointer dereference at virtual address 00000ef8
[   37.683157] pgd = 1377b4d8
[   37.685908] [00000ef8] *pgd=13294003, *pmd=00000000
[   37.690882] Internal error: Oops: 206 [#1] PREEMPT SMP ARM
[   37.696454] Modules linked in:
[   37.699562] CPU: 2 PID: 683 Comm: wpa_supplicant Not tainted 5.4.58-v7l-g6c0b7b018b84 #1
[   37.707772] Hardware name: BCM2711
[   37.711246] PC is at brcmf_p2p_send_action_frame+0x3e8/0x9c0
[   37.716994] LR is at brcmf_p2p_send_action_frame+0x19c/0x9c0
[   37.722740] pc : [<c08ebe68>]    lr : [<c08ebc1c>]    psr: 20010013
[   37.729108] sp : d432fad0  ip : 00000000  fp : d432fb24
[   37.734414] r10: 00000000  r9 : ffffffff  r8 : 00000000
[   37.739720] r7 : d3c68800  r6 : 00000001  r5 : d4f0a80c  r4 : d4f0a800
[   37.746345] r3 : 00000400  r2 : 00000000  r1 : 00000006  r0 : 00000000
[   37.752965] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   37.760212] Control: 30c5383d  Table: 13a2a740  DAC: 55555555
[   37.766041] Process wpa_supplicant (pid: 683, stack limit = 0xf9b8e57a)
[   37.772745] Stack: (0xd432fad0 to 0xd4330000)
[   37.777159] fac0:                                     00000304 c04356b8 d432fb24 d432fae8
[   37.785459] fae0: ef803540 c043599c d432fb88 d4fb0620 d4f0b640 00000000 d432fb24 c1806448
[   37.793759] fb00: d41aac34 d432fbb8 d4f0a800 0000002f 00000017 d3c68800 d432fb74 d432fb28
[   37.802057] fb20: c08e1d0c c08eba8c 00000000 00000000 00000017 00000985 d4fb0620 d4ea1008
[   37.810350] fb40: 00000985 5cac72ff d432fb84 d41aac34 d4ea1008 d432fbe4 d4fb0280 d432fbb8
[   37.818643] fb60: d4fb0000 d3a70014 d432fbac d432fb78 c0e597e4 c08e1b64 d432fbac d432fb88
[   37.826941] fb80: c0c6ddac d432fc74 c1806448 d4fb0000 d4ea1008 00000000 d49e9300 d3a70014
[   37.835242] fba0: d432fc34 d432fbb0 c0e4ee60 c0e596dc 00000000 0000003b 00000000 00000000
[   37.843536] fbc0: c115ec34 00000000 c10396f8 d4f06d18 00000000 00000985 00000000 00000000
[   37.851827] fbe0: 00000000 d4f06d18 00000001 000003e8 d41aac34 0000002f 00000000 00000000
[   37.860126] fc00: 00000000 5cac72ff c070cf24 c115fba0 c1412e80 d41aac14 c1806448 d49e9000
[   37.868417] fc20: 00000000 c1895ec0 d432fccc d432fc38 c0c6ea64 c0e4ebf0 c115ea9c 00000000
[   37.876716] fc40: d432fcd4 00000004 c196e258 00000014 d3c6b000 d432fcd4 d432fcdc d432fc68
[   37.885017] fc60: c065488c c02b1dc0 d39ce000 d3348d30 00000164 5f352d63 570002ab d41aac00
[   37.893318] fc80: d41aac10 d41aac14 d3c6b000 c1895ec0 d4fb0000 d4ea1008 d432fcd4 5cac72ff
[   37.901611] fca0: efa50980 00000000 c1806448 c0c6e7e0 d49e9000 d41aac00 00000000 efb57800
[   37.909902] fcc0: d432fd1c d432fcd0 c0c6d774 c0c6e7ec 032c0000 00000000 00000000 00000000
[   37.918194] fce0: 00000000 00000000 00000000 00000000 00000000 5cac72ff 00000000 c189ab50
[   37.926492] fd00: d49e9000 efb57864 00000064 c1895ec0 d432fd34 d432fd20 c0c6de40 c0c6d688
[   37.934792] fd20: efa08240 d49e9000 d432fd74 d432fd38 c0c6ce70 c0c6de18 00000000 c1806448
[   37.943084] fd40: 7fffffff 5cac72ff d432fd74 c1806448 d49e9000 d3823c00 00000000 00000000
[   37.951375] fd60: d432ff4c 00000000 d432fde4 d432fd78 c0c6d1e0 c0c6cc88 00000000 00000000
[   37.959668] fd80: 00000000 00000000 00000000 00000000 d41aac00 00000064 c14124e0 d42db7c0
[   37.967965] fda0: 00000000 000002ab 000003f2 000003f2 00000000 5cac72ff d432fde4 d432ff4c
[   37.976261] fdc0: 00000000 00000000 c1806448 d432fdf4 d4701080 00000000 d432fe44 d432fde8
[   37.984554] fde0: c0bdcf80 c0c6cf0c d432e000 00000000 d432fe44 d432fe00 c0bdd1cc c06e2760
[   37.992845] fe00: d432fe50 d432ff54 b3335cf4 0000000c beff7410 5cac72ff 00000000 c1806448
[   38.001135] fe20: d432ff4c 00000000 d4701080 00000000 d432e000 00000128 d432ff3c d432fe48
[   38.009428] fe40: c0bdd2cc c0bdcda0 00000000 c06547fc 00000000 b3346000 00000064 c0bdcbac
[   38.017719] fe60: 00008000 00000000 d432fe80 5cac72ff 00039e7e 00000000 ffffffff 00000001
[   38.026011] fe80: 00000000 5cac72ff c0bdcbac c1806448 d31e9f00 00000010 00000000 00000000
[   38.034302] fea0: d4701080 00000126 d432ff24 c03a11c4 c1940ca8 c1806448 c181c110 d432ff68
[   38.042594] fec0: d3823c00 beff7468 d432ff64 00000001 d432ff54 d432fee0 c03a11c4 c02b1dc0
[... 104 lines cut ...]
[   38.764121] 3fc0: a4ca8eb0 a4cb7040 a4ca8ed4 00000006 00000000 00000017 a4c2f368 a4cb77a8
[   38.772412] 3fe0: 00000000 9ec043d8 a31c0ced a31c0d90 200d0030 ffffffff
[   38.779117]  r8:30c5387d r7:30c5383d r6:ffffffff r5:200d0030 r4:a31c0d90 r3:a4cb78f0
[   38.786967] CPU0: stopping
[   38.789709] CPU: 0 PID: 231 Comm: surfaceflinger Tainted: G      D           5.4.58-v7l-g6c0b7b018b84 #1
[   38.799321] Hardware name: BCM2711
[   38.802764] Backtrace: 
[   38.805242] [<c020dbd4>] (dump_backtrace) from [<c020ded0>] (show_stack+0x20/0x24)
[   38.812917]  r6:00000000 r5:c18a9a0c r4:600f0193 r3:5cac72ff
[   38.818653] [<c020deb0>] (show_stack) from [<c0e8baa0>] (dump_stack+0xd0/0x114)
[   38.826064] [<c0e8b9d0>] (dump_stack) from [<c0212248>] (handle_IPI+0x3a8/0x424)
[   38.833563]  r9:c1806bb4 r8:f0815000 r7:00000000 r6:c18bef60 r5:c16c2950 r4:00000004
[   38.841415] [<c0211ea0>] (handle_IPI) from [<c0202310>] (gic_handle_irq+0x84/0x88)
[   38.849089]  r10:00000000 r9:b06df810 r8:f0815000 r7:f0814000 r6:d3e27fb0 r5:f081400c
[   38.857027]  r4:c1806ef4
[   38.859591] [<c020228c>] (gic_handle_irq) from [<c0201dac>] (__irq_usr+0x4c/0x60)
[   38.867176] Exception stack(0xd3e27fb0 to 0xd3e27ff8)
[... cut ...]
[   38.880587] 7fc0: b06a91a0 b06df928 b061d6f8 00000000 b06dfb04 b06df810 00000000 b2621668
[   38.888878] 7fe0: b26e07bc bef21e18 b26492c9 b2e58eba 600f0030 ffffffff
[   38.895583]  r8:30c5387d r7:30c5383d r6:ffffffff r5:600f0030 r4:b2e58eba r3:00000004
[   38.903444] Rebooting in 5 seconds..
------------------------------------------------------------------------------------------

It seems that the wlan supplant has something wrong,would you build a room **without the wifi enabled by default**? Then I cam debug it and report the result to you,thanks!

KonstaT, 2020-08-16:
Thanks for the log. This seems like the issue discussed before. http://disq.us/p/2ax3vmu

You can disable wifi by adding dtoverlay=disable-wifi to /boot/config.txt.

andersling, 2020-08-16:
[KernelDBG_Msg_DisableWIFI_Room20200813ByKonstaT.txt](https://github.com/lineage-rpi/android_kernel_brcm_rpi/files/5080769/KernelDBG_Msg_DisableWIFI_Room20200813ByKonstaT.txt)
Thanks for you reply, and just now I try to disabled wifi follow your tips, and IT CAN BOOT UP NOW! That is really exciting!!! Thanks ,sir!
   Hope in the new version that you can fixed the wifi isue, big thanks!

andersling, 2020-08-16:
and by the way, can i enable the ethernet interface so that i can use it to connect to the internet.. Thanks!

andersling, 2020-08-16:
anther information report to you: when the room boot up successfully, i try to turn the wifi on in the setting ui, after a few seconds, the on went back to off automatic, and the USB debug message show no new messages! and after that ,the system is still runing normall ,NO Reboots!
   hope this can be referred by you, thanks!

KonstaT, 2020-08-17:
I can't reproduce this issue myself which makes testing possible fixes really difficult.

Ethernet is automatically enabled when you plug in a cable. There's no settings options for it but you'll see an icon on the status bar when you're connected using ethernet.

poudaduk, 2020-10-22:
Hello,

Firstly, like many others, I'd like to offer my sincere thanks for your work.  It is really fantastic.  I am having the same rebooting loop issue, and I did see that you've not been able to reproduce it, but I am hoping by giving some more symptoms / data points, it might be useful.

I have a Pi 4 with 8GB RAM (first one).  I initially experienced the exact symptoms described above (would get just past the LineageOS splash screen then reboot).  I followed the directions, and this is where it gets a little interesting.  I changed the config.txt file to shut the Wi-Fi off.  It booted, but only got to the screen "Enter Name of Wi-Fi network" - I couldn't get past that.  I tried this with both the standard LineageOS and AndroidTV builds with the same result.

Then, I reflashed the card with both builds, but did not change the config.txt file , I plugged the Ethernet cable in.  It worked.   Once plugged in, I tried to join a Wi-Fi network - it reboots.  If I turn off Wi-Fi and stay on Ethernet, it works fine.

Ultimately, I'd like to use Wi-Fi, and I am sure that will happen.  Until then, I sincerely hope this provides a little more data.  I can also add I have tried two different micro SD cards - same result.  I have also installed Raspberry Pi OS as a test and it works fine on all fronts.

Thanks again for your time and work.

andersling, 2020-10-22:
> Hello,
>
> Firstly, like many others, I'd like to offer my sincere thanks for your work. It is really fantastic. I am having the same rebooting loop issue, and I did see that you've not been able to [...]

Hi, The main issue is the wifi-driver not works!  There is no solution yet, what we have to is waiting MR.Kansta release some fix.
 and by the way ,you can use the  Android-RPI 11 (64bit), On all my 2G/4G/8G raspberryPI4 boards, it works well including wifi and bluetooth.  https://github.com/android-rpi/device_arpi_rpi4 

Have a good day!

Dushyantc1, 2020-10-26:
> Hi,sir! When I use you 20200813 rom for pi4,it keep rebooting and rebooting, i catch the debug messages from USB ,please take a look at it,thanks!
> 
> ## [ 35.187724] type=1400 audit(1597320517.289:101): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 35.200637] init: Could not find 'vendor.lineage.livedisplay@2.0::IAutoContrast/default' for ctl.interface_start
> [ 35.219864] type=1400 audit(1597320517.289:101): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 35.230787] init: Received control message 'interface_start' for 'vendor.lineage.livedisplay@2.0::IDisplayColorCalibration/default' from pid: 146 (/system/bin/hwservicemanager)
> [ 35.247169] type=1400 audit(1597320517.289:102): avc: denied { open } for comm="main" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 35.263288] init: Could not find 'vendor.lineage.livedisplay@2.0::IDisplayColorCalibration/default' for ctl.interface_start
> [ 35.294351] init: Received control message 'interface_start' for 'vendor.lineage.livedisplay@2.0::IDisplayModes/default' from pid: 146 (/system/bin/hwservicemanager)
> [ 35.309852] init: Could not find 'vendor.lineage.livedisplay@2.0::IDisplayModes/default' for ctl.interface_start
> [ 35.320421] init: Received control message 'interface_start' for 'vendor.lineage.livedisplay@2.0::IReadingEnhancement/default' from pid: 146 (/system/bin/hwservicemanager)
> [ 36.851865] type=1400 audit(1597320517.289:102): avc: denied { open } for comm="main" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 36.873786] type=1400 audit(1597320518.979:103): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 36.894704] type=1400 audit(1597320518.979:103): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 36.915905] type=1400 audit(1597320518.979:104): avc: denied { open } for comm="main" path="/dev/stune/top-app/tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 37.175658] type=1400 audit(1597320518.979:104): avc: denied { open } for comm="main" path="/dev/stune/top-app/tasks" dev="tmpfs" ino=731 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
> [ 37.198532] type=1400 audit(1597320519.309:105): avc: denied { write } for comm="main" name="tasks" dev="tmpfs" ino=10663 scontext=u:r:zygote:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1 app=com.android.providers.media
> [ 37.670057] 8<--- cut here ---
> [ 37.674677] Unable to handle kernel NULL pointer dereference at virtual address 00000ef8
> [ 37.683157] pgd = 1377b4d8
> [ 37.685908] [00000ef8] *pgd=13294003, *pmd=00000000
> [ 37.690882] Internal error: Oops: 206 [#1] PREEMPT SMP ARM
> [ 37.696454] Modules linked in:
> [ 37.699562] CPU: 2 PID: 683 Comm: wpa_supplicant Not tainted 5.4.58-v7l-g6c0b7b018b84 #1
> [ 37.707772] Hardware name: BCM2711
> [ 37.711246] PC is at brcmf_p2p_send_action_frame+0x3e8/0x9c0
> [ 37.716994] LR is at brcmf_p2p_send_action_frame+0x19c/0x9c0
> [ 37.722740] pc : [] lr : [] psr: 20010013
> [ 37.729108] sp : d432fad0 ip : 00000000 fp : d432fb24
> [ 37.734414] r10: 00000000 r9 : ffffffff r8 : 00000000
> [ 37.739720] r7 : d3c68800 r6 : 00000001 r5 : d4f0a80c r4 : d4f0a800
> [ 37.746345] r3 : 00000400 r2 : 00000000 r1 : 00000006 r0 : 00000000
> [ 37.752965] Flags: nzCv IRQs on FIQs on Mode SVC_32 ISA ARM Segment user
> [ 37.760212] Control: 30c5383d Table: 13a2a740 DAC: 55555555
> [ 37.766041] Process wpa_supplicant (pid: 683, stack limit = 0xf9b8e57a)
> [ 37.772745] Stack: (0xd432fad0 to 0xd4330000)
> [ 37.777159] fac0: 00000304 c04356b8 d432fb24 d432fae8
> [ 37.785459] fae0: ef803540 c043599c d432fb88 d4fb0620 d4f0b640 00000000 d432fb24 c1806448
> [ 37.793759] fb00: d41aac34 d432fbb8 d4f0a800 0000002f 00000017 d3c68800 d432fb74 d432fb28
> [ 37.802057] fb20: c08e1d0c c08eba8c 00000000 00000000 00000017 00000985 d4fb0620 d4ea1008
> [ 37.810350] fb40: 00000985 5cac72ff d432fb84 d41aac34 d4ea1008 d432fbe4 d4fb0280 d432fbb8
> [ 37.818643] fb60: d4fb0000 d3a70014 d432fbac d432fb78 c0e597e4 c08e1b64 d432fbac d432fb88
> [ 37.826941] fb80: c0c6ddac d432fc74 c1806448 d4fb0000 d4ea1008 00000000 d49e9300 d3a70014
> [ 37.835242] fba0: d432fc34 d432fbb0 c0e4ee60 c0e596dc 00000000 0000003b 00000000 00000000
> [ 37.843536] fbc0: c115ec34 00000000 c10396f8 d4f06d18 00000000 00000985 00000000 00000000
> [ 37.851827] fbe0: 00000000 d4f06d18 00000001 000003e8 d41aac34 0000002f 00000000 00000000
> [ 37.860126] fc00: 00000000 5cac72ff c070cf24 c115fba0 c1412e80 d41aac14 c1806448 d49e9000
> [ 37.868417] fc20: 00000000 c1895ec0 d432fccc d432fc38 c0c6ea64 c0e4ebf0 c115ea9c 00000000
> [ 37.876716] fc40: d432fcd4 00000004 c196e258 00000014 d3c6b000 d432fcd4 d432fcdc d432fc68
> [ 37.885017] fc60: c065488c c02b1dc0 d39ce000 d3348d30 00000164 5f352d63 570002ab d41aac00
> [ 37.893318] fc80: d41aac10 d41aac14 d3c6b000 c1895ec0 d4fb0000 d4ea1008 d432fcd4 5cac72ff
[... cut ...]
> [ 38.888878] 7fe0: b26e07bc bef21e18 b26492c9 b2e58eba 600f0030 ffffffff
> [ 38.895583] r8:30c5387d r7:30c5383d r6:ffffffff r5:600f0030 r4:b2e58eba r3:00000004
> [ 38.903444] Rebooting in 5 seconds..
> It seems that the wlan supplant has something wrong,would you build a room **without the wifi enabled by default**? Then I cam debug it and report the result to you,thanks!

Hello, How did you catch this debug on USB. I am having the same issue with Konstakang's buid. I want to catch the log whether what is going wrong. I do not know how to get the log. Please help.

Dushyantc1, 2020-10-26:
> Thanks for the log. This seems like the issue discussed before (http://disq.us/p/2ax3vmu). I still have no idea why this only happens on some devices.
> 
> You can disable wifi by adding dtoverlay=disable-wifi to /boot/config.txt.

I disabled the wifi, It successfully boot up but it now stuck on Select your Wifi Network. It is not allowing me to skip the wifi connection screen. I have inserted the LAN cable, still it is not connecting to internet, I mean still it is not allowing me to bypass the wifi connection screen

andersling, 2020-10-27:
> I disabled the wifi, It successfully boot up but it now stuck on Select your Wifi Network. It is not allowing me to skip the wifi connection screen. I have inserted the LAN cable, still it is not [...]

Are you sure you disabled WIFI ? As for me ,the LAN will auto connect by DHCP .

andersling, 2020-10-27:
> Hello, How did you catch this debug on USB. I am having the same issue with Konstakang's buid. I want to catch the log whether what is going wrong. I do not know how to get the log. Please help.

Hi, you can search youtube or google by "Debug kernel Message in RaspberryPI" , you will find a lot useful instruction to do it. 
First of all ,you have to prepare a USB To TTL adapter .maybe it cast 3-5 dollars in amazon.
Have a nice day!

and btw, you can try android-rpi 11  64bit room, it is all works with wifi and bluetooth.

KonstaT, 2020-10-27:
As I suspected elsewhere wifi network surroundings seem to have an effect on this (http://disq.us/p/2cj20pe). It would be interesting to know which device nearby you need to turn on/off to be able to reproduce this.

Also could someone run 'iogcat' and post the output as well? I've only seen the kernel log so far.

andersling, 2020-10-30:
> As I suspected elsewhere wifi network surroundings seem to have an effect on this (http://disq.us/p/2cj20pe). It would be interesting to know which device nearby you need to turn on/off to be able [...]

Hi,MR. KonstaT! Do you mean that the Device Driver decode the WIFI Beacon packets,some ap's beacons are no so good ,such as 802.11ac or 802.11ax ,or the vendor Information Elements( IEs)  ..etc, which cause the driver decode error?
I am familar with the WIFI standard such as 802.11n, ac ,ax. how can i do something for you? capture the beacon frames with wireshark format or elsething? 
   Have a nice day,sir!

KonstaT, 2020-10-30:
> Hi,MR. KonstaT! Do you mean that the Device Driver decode the WIFI Beacon packets,some ap's beacons are no so good ,such as 802.11ac or 802.11ax ,or the vendor Information Elements( IEs) ..etc, [...]

No, I'm not interested about packet captures. If you could run 'logcat' using serial console adapter that might bring some new information.

From practical point of view I'd more interested to know what kind of device I'd need to bring to my network surroundings so I could reproduce this issue myself. Can you turn off some surrounding devices. Does this only happen after turning some device on? E.g. wireless CarlinKit adapter seems to be one this can reproduced with (https://www.raspberrypi.org/forums/viewtopic.php?f=73&t=281187&sid=06c476372818de3df141ee7dbf7c6238&start=150#p1749890) but I'm not going to spend ~100€ on something I have no other use for.

govindmaheshwari2, 2020-11-06:
I have tried it by switching off different bandwidth wifi and i found that with 5ghz bandwidth it get in rebooting loop with other wifi it work fine.

KonstaT, 2020-11-11:
@govindmaheshwari2
What makes you think this is somehow related 5GHz?

5GHz networks are not normally visible on my device for some reason when my wifi region is set to FI. If I set wifi region to US (iw reg set US) I can connect to my 5GHZ AP fine. No reboots.

ratuozora, 2020-11-14:
Hi Mr. Konsta,

Your build is working great! Thanks A Lot. 

I have experienced with wifi issue like the others, it could be not enough power on rpi4, so it reboot. 
I did booting LOS 17.1 with 5 volt and 1 amp power supply, but when I turn on the wifi , it reboot. 
So I tried with greater power supply 5 volt and 2.4 amp and no problem, run perfectly. 
Actually recommended from the official rpi4 is 5 v and 3 Amp.
so I think that the wifi device consume much power and can cause the rpi4 processors under voltage or power that make it reboot.
I have experienced this under power symtoms since rpi1 and other mobile device. I hope this can help.

Happy LineAgeOS!!!

KonstaT, 2020-11-20:
Please test the latest LineageOS 17.1 builds on the Pi 4 and report if there's any change. I patched the kernel around where this is crashing.

govindmaheshwari2, 2020-11-20:
No change still the same issue. Rebooting again again

KonstaT, 2020-11-20:
@govindmaheshwari Which build did you use?

govindmaheshwari2, 2020-11-20:
The one you release yesterday for android tv lineage-17.1-20201119

KonstaT, 2020-11-20:
Ok, thanks for the report. 

I guess I still need to find some way to reproduce this myself. And I'm also yet to even see a logcat of this on LineageOS.

govindmaheshwari2, 2020-11-20:
I can share my screen and device so that you can see and reproduce these issue like teamviewer or any other required software. So that you can make it correct and you don't have to buy expensive things

KonstaT, 2020-11-20:
@govindmaheshwari2 No, that's doesn't work in any for something like this.

UMB8998, 2020-11-23:
@KonstaT @govindmaheshwari2 I also have the same problem, when I use the raspberry with wifi it goes into a reboot loop, even with a very powerful powersupply, the issue happens with 2.4 and 5 ghz wifi.
I will get a Serial Adapter and run more tests on this
BTW it is a great work, Thank you!!!!!

govindmaheshwari2, 2021-01-07:
@KonstaT your new build lineage-18.1-20210104-UNOFFICIAL-KonstaKANG-rpi4 is working great. there is no issue of wifi or any other thing i can find right now. Can you please provide TV version of lineage 18.1. GAPPS is not an issue there are lots of 3rd party store can be used. Thank you for building a great and smooth working android 11 rom.

KonstaT, 2021-01-07:
@govindmaheshwari2 Thanks for the report. There's no plans for Android TV 11 releases in any near future (it does not offer anything you don't already have with current Android TV builds). You can install a different launcher if you want to have a more TV friendly UI.

Difference between LineageOS 17.1 and 18.1 kernels is just one configuration option so it looks like the root of this issue was not in the kernel after all. Like I suspected this is fixed somewhere else in Android 11 most likely wpa_supplicant. I would still like to have this fixed for older Android versions of course.

KonstaT, 2021-01-26:
There's also a report that this issue is not resolved with LineageOS 18. 

It seems that this is somehow related to Hotspot 2.0/802.11u/Access Network Query Protocol (ANQP) networks being present in the network environment.

```
2021-01-24 17:39:39.259 513-513/? D/wpa_supplicant: nl80211: Drv Event 34 (NL80211_CMD_NEW_SCAN_RESULTS) received for wlan0
2021-01-24 17:39:39.259 513-513/? D/wpa_supplicant: wlan0: nl80211: New scan results available
2021-01-24 17:39:39.259 513-513/? D/wpa_supplicant: nl80211: Scan probed for SSID ''
2021-01-24 17:39:39.259 513-513/? D/wpa_supplicant: nl80211: Scan included frequencies: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700
2021-01-24 17:39:39.259 513-513/? D/wpa_supplicant: wlan0: Event SCAN_RESULTS (3) received
2021-01-24 17:39:39.259 513-513/? D/wpa_supplicant: nl80211: Received scan results (6 BSSes)
2021-01-24 17:39:39.259 384-765/system_process D/WifiNl80211Manager: Scan result ready event
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: BSS: Start scan result update 1
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: BSS: Add new id 0 BSSID a0:39:ee:8a:b0:16 SSID 'LinkSys' freq 2437
2021-01-24 17:39:39.260 384-765/system_process D/WifiNative: Scan result ready event
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: BSS: Add new id 1 BSSID a2:39:ee:8a:b0:17 SSID 'LinkGuest' freq 2437
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: BSS: Add new id 2 BSSID e0:ce:c3:c7:ee:15 SSID 'AKADO-0040' freq 2462
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: BSS: Add new id 3 BSSID ec:41:18:f5:f3:d9 SSID 'MIEPL REKTOR' freq 2417
2021-01-24 17:39:39.260 384-401/system_process I/EthernetTracker: interfaceLinkStateChanged, iface: wlan0, up: false
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: BSS: Add new id 4 BSSID e1:8c:2a:99:b5:92 SSID 'Hotspot 2.0' freq 2442 HESSID 2c:c5:d3:11:50:90
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: BSS: Add new id 5 BSSID a0:8e:78:01:c0:1e SSID 'AKADO-C018' freq 2412
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: BSS: last_scan_res_used=6/32
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: New scan results available (own=0 ext=1)
2021-01-24 17:39:39.260 513-513/? D/wpa_supplicant: wlan0: Do not use results from externally requested scan operation for network selection
2021-01-24 17:39:39.261 513-513/? D/wpa_supplicant: RTM_NEWLINK: ifi_index=8 ifname=wlan0 wext ifi_family=0 ifi_flags=0x1003 ([UP])
2021-01-24 17:39:39.263 384-467/system_process D/WifiNl80211Manager: get 6 scan results from wificond
2021-01-24 17:39:39.277 384-467/system_process D/WifiNative: get 6 scan results from wificond
2021-01-24 17:39:39.278 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159764
2021-01-24 17:39:39.283 384-467/system_process I/chatty: uid=1000(system) WifiScanningSer identical 22 lines
2021-01-24 17:39:39.283 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159764
2021-01-24 17:39:39.284 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159749
2021-01-24 17:39:39.285 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159767
2021-01-24 17:39:39.286 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159749
2021-01-24 17:39:39.286 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159767
2021-01-24 17:39:39.287 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159749
2021-01-24 17:39:39.287 384-464/system_process D/WifiScanRequestProxy: Scan results received
2021-01-24 17:39:39.287 384-464/system_process D/WifiScanRequestProxy: Received 6 scan results
2021-01-24 17:39:39.287 384-467/system_process D/WifiAsyncChannel.WifiScanningService: sendMessage sendMessage message=159749
2021-01-24 17:39:39.288 384-505/system_process D/WifiHealthMonitor:  2G scanResult count: 6, Above2g scanResult count: 0
2021-01-24 17:39:39.291 384-1401/system_process I/WifiService: getScanResults uid=1000
2021-01-24 17:39:39.295 384-1401/system_process I/WifiService: getWifiEnabledState uid=1000
2021-01-24 17:39:39.295 384-1401/system_process I/WifiService: getScanResults uid=1000
2021-01-24 17:39:39.298 384-1401/system_process I/WifiService: getSupportedFeatures uid=1000
2021-01-24 17:39:39.298 384-1401/system_process D/WifiAsyncChannel.WifiService: sendMessageSynchronously sendMessageSynchronously.send message=131133
2021-01-24 17:39:39.298 384-464/system_process D/WifiClientModeImpl:  DefaultState screen=on uid=1000 0 0
2021-01-24 17:39:39.299 384-1401/system_process D/WifiAsyncChannel.WifiService: sendMessageSynchronously sendMessageSynchronously.recv message=131133 sendingUid=1000
2021-01-24 17:39:39.308 384-1401/system_process I/WifiService: getSupportedFeatures uid=1000
2021-01-24 17:39:39.308 384-1401/system_process D/WifiAsyncChannel.WifiService: sendMessageSynchronously sendMessageSynchronously.send message=131133
2021-01-24 17:39:39.308 384-464/system_process D/WifiClientModeImpl:  DefaultState screen=on uid=1000 0 0
[... cut ...]
2021-01-24 17:39:39.369 384-464/system_process D/HS20: ANQP initiated on e1:8c:2a:99:b5:92
2021-01-24 17:39:39.370 384-464/system_process D/PasspointManager: ANQP entry not found for: e1:8c:2a:99:b5:92:<Hotspot 2.0>
2021-01-24 17:39:39.374 384-1401/system_process I/WifiService: getMatchingOsuProviders uid=1000
2021-01-24 17:39:39.375 384-1401/system_process I/WifiService: getMatchingPasspointConfigsForOsuProviders uid=1000
2021-01-24 17:39:39.378 384-1401/system_process I/WifiService: getWifiEnabledState uid=1000
reboot
```

Rpi4Alex, 2021-01-26:
Konsta Hi.
I've just noticed a very confusing thing.  I changed my box from standard one (https://ae01.alicdn.com/kf/H7b384e742d274837b315fce47bb389134.jpg_q50.jpg) to passive one (https://http2.mlstatic.com/D_NQ_NP_830683-MLM31962423570_082019-O.webp) and it stopped rebooting. I keep on looking afer its behavior...   24hrs of stable work so far.

KonstaT, 2021-01-27:
@Rpi4Alex No, unlikely that case anything to do with this. This is to do with specific type of wifi network being in your surroundings. In your case it was the BSSID e1:8c:2a:99:b5:92 SSID 'Hotspot 2.0' network which is most likely now just turned off.

Rpi4Alex, 2021-01-28:
Konsta, Hi.
Thanks once again for your quick feedbacks. It is per se worth a lot.
No,... You are right . The new case doesn't resolve the issue. It still reboots. Herewith i attach the latest log but it seems to me that it doesn'r show anything since the device was connected via wifi ( https://yadi.sk/d/BWtKkfwRCN0D1g ). 
I will try to connect it via usb (waiting for the device) later on.

KonstaT, 2021-01-28:
BSSID e1:8c:2a:99:b5:92 SSID 'Hotspot 2.0' is there on fullLogBeforeReboot4.txt at least. Is that your or some neighbors network? Try switching it off if you can.

Rpi4Alex, 2021-01-28:
I dont even see this network in the list of availabele wifi networks.
Is there a way to block the specified network (or number of networks) from detection (exclude it from the availability list) and stick to one predefined by user?

KonstaT, 2021-01-28:
@Rpi4Alex No, I'm not aware of any way you could that. This issue needs some proper way to resolve it anyway.

Rpi4Alex, 2021-01-28:
Just to let you know if it usefull: all android OS - not only LOS (at least those recently installed on my device) suffered from the same bug. 
PaspOS doesnt have such an issue where i believe the environment (or something else) is the same. Maybe here is the clue...

KonstaT, 2021-01-28:
Never even heard of PaspOS and apparently neither has Google search.

There seems to be mixed reports with this as people have reported that this is no longer an issue with Android 11 builds.

Rpi4Alex, 2021-01-29:
Soory, didnt notice this typo. It should have been RaspOS not PaspOS.

doolijb, 2021-02-20:
Concur, having this issue as well at a complex with lots of wifi signals.

msugirin, 2021-03-02:
And I thought I was the only one in the boat - I even bought a new RP4 with more memory just to isolate it's not a faulty RP4. I tried 2 different RP4s (2gb and 8gb) model and 3 different SD cards and all have the same reboot problem.

Wifi disabled - all looks good.

Here's a logcat right before it rebooted, hopefully it helps?

`03-01 21:03:32.135   702  1803 W TileUtils: Found com.android.settings.Settings$DataUsageSummaryActivity for intent Intent { act=com.android.settings.action.SETTINGS pkg=com.android.settings } missing metadata com.android.settings.category
03-01 21:03:32.150   386   893 I WifiService: startScan uid=1000
03-01 21:03:32.162   702   702 W ConnectivityManager.CallbackHandler: callback not found for CALLBACK_AVAILABLE message
03-01 21:03:32.168   386   902 D PackageManager: Instant App installer not found with android.intent.action.INSTALL_INSTANT_APP_PACKAGE
03-01 21:03:32.169   386   902 D PackageManager: Clear ephemeral installer activity
03-01 21:03:32.176   702  1803 D SettingsActivity: Enabled state changed for some tiles, reloading all categories {com.android.settings/com.android.settings.Settings$PowerUsageSummaryActivity},
03-01 21:03:32.181   702  1667 W TileUtils: Found com.android.settings.Settings$DataUsageSummaryActivity for intent Intent { act=com.android.settings.action.SETTINGS pkg=com.android.settings } missing metadata com.android.settings.category
03-01 21:03:32.596   386   496 D ConnectivityService: releasing NetworkRequest [ TRACK_DEFAULT id=18, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED Uid: 1000] ] (release request)
03-01 21:03:32.600   562   562 V KeyguardUpdateMonitor: onSubscriptionInfoChanged()
03-01 21:03:33.161   386   413 E system_server: No package ID 7f found for ID 0x7f080249.
03-01 21:03:33.161   386   413 E system_server: No package ID 7f found for ID 0x7f121205.
03-01 21:03:33.161   386   413 E system_server: No package ID 7f found for ID 0x7f080248.
03-01 21:03:33.161   386   413 E system_server: No package ID 7f found for ID 0x7f120517.
03-01 21:03:33.162   386   413 E system_server: No package ID 7f found for ID 0x7f080247.
03-01 21:03:33.162   386   413 E system_server: No package ID 7f found for ID 0x7f120b34.
03-01 21:03:33.168   386   482 I InputReader: Reconfiguring input devices.  changes=0x00000010
03-01 21:03:33.172   702  1667 W TileUtils: Found com.android.settings.Settings$DataUsageSummaryActivity for intent Intent { act=com.android.settings.action.SETTINGS pkg=com.android.settings } missing metadata com.android.settings.category
03-01 21:03:33.176   386   386 I Telecom : DefaultDialerCache: Refreshing default dialer for user 0: now null: DDC.oR@AA8
03-01 21:03:33.178   386   413 W Searchables: No global search activity found
03-01 21:03:33.180   386   413 W VoiceInteractionManagerService: no available voice interaction services found for user 0
03-01 21:03:34.174   386   893 D WificondControl: Scan result ready event
03-01 21:03:34.179   386   495 D WificondScannerImpl: Filtering out 2 scan results.
03-01 21:03:34.180   386   405 I EthernetTracker: interfaceLinkStateChanged, iface: wlan0, up: true
03-01 21:03:34.181   202   202 I netd    : interfaceGetCfg("wlan0") <0.79ms>
03-01 21:03:34.187   636   636 I wpa_supplicant: wlan0: GAS-QUERY-START addr=ee:22:80:12:75:fd dialog_token=0 freq=5240
03-01 21:03:34.187   386   493 D HS20    : ANQP initiated on ee:22:80:12:75:fd
03-01 21:03:34.188   386   493 D PasspointManager: ANQP entry not found for: ee:22:80:12:75:fd:<AlticeWiFi>
03-01 21:03:34.212   702   702 D WifiSettings: onAccessPointsChanged (WifiTracker) callback initiated`

msugirin, 2021-03-02:
I did some more additional digging to see what this Wifi actually is, and it seems to be a public hotspot from a Cable Company in the area.

Is it maybe related to that?

KonstaT, 2021-03-02:
@msugirin Yes, as commented before this seems to be related to Hotspot 2.0/802.11u/Access Network Query Protocol (ANQP) networks being present in the network environment. BSSID ee:22:80:12:75:fd SSID 'AlticeWiFi' in your case.

Which LineageOS version is this BTW? There's been some mixed reports about this being fixed in LineageOS 18.

msugirin, 2021-03-02:
This is on the January release of 17.1 (lineage-17.1-20210122). I tried the same on the 18.1 (lineage-18.1-20210214) release and ran into the same issue during the setup screen. 

I can retry it on 18.1 and also check the logcats - will post later

msugirin, 2021-03-02:
Same issue on latest 18.1 build:

`03-02 08:34:10.929   391  1760 D WifiNl80211Manager: Scan result ready event
03-02 08:34:10.929   391  1760 D WifiNative: Scan result ready event
03-02 08:34:10.957   391   471 D PasspointManager: ANQP entry not found for: ee:22:80:12:75:fd:<AlticeWiFi>
03-02 08:34:10.962   520   520 I wpa_supplicant: wlan0: GAS-QUERY-START addr=ee:22:80:12:75:fd dialog_token=165 freq=5240
03-02 08:34:10.963   391   471 D HS20    : ANQP initiated on ee:22:80:12:75:fd
03-02 08:34:10.963   391   471 D PasspointManager: ANQP entry not found for: ee:22:80:12:75:fd:<AlticeWiFi>
03-02 08:34:11.014   391   471 D ANQPRequestManager: Not allowed to send ANQP request to ee:22:80:12:75:fd for another 9 seconds
03-02 08:34:11.015   391   471 D PasspointManager: ANQP entry not found for: ee:22:80:12:75:fd:<AlticeWiFi>
12-31 13:14:05.477     0     0 F         : 8<--- cut here ---
12-31 13:14:05.480     0     0 F         : Unable to handle kernel NULL pointer dereference at virtual address 00000000
03-02 08:34:11.029   391   442 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
`

NumberOneGit, 2021-04-05:
Same issue here.  It looks like RPi and Android won't be an option for a lot of us in densely populated areas until this is resolved.  I have the same issue wherever I try to boot up in NYC.

matthuo333, 2021-04-26:
Is there any conclusion on this question?

matthuo333, 2021-04-26:
How to set to skip Ethernet detection? Because I am in China, I need a VPN, but there is no place to set the gateway address during the detection. After entering the system, I need to manually set the network

yashsurya, 2021-07-29:
is it fixed yet? i am using raspi 3b+ with 5v 3amp power supply and it only works when i disable wifi with the config.txt file .

KonstaT, 2021-12-16:
This issue should be now resolved in one way with the latest LineageOS 19 build (https://konstakang.com/devices/rpi4/LineageOS19/). Please test and confirm that this is the case.

I'll also include similar fix to LineageOS 18 builds if/when there's another release.

msugirin, 2021-12-27:
Looks promising so far! Just flashed the image (20211216) and seems to be holding. I tried earlier with (19.0.20211125), but that had also reboot issues.

Kept it on Wifi scanning for about 5 minutes so far and no reboots.

msugirin, 2021-12-28:
Ran it all day, used different apps, downloaded/uninstalled apps. Not a single reboot :) so happy!

Now I can finally re-start my car-droid project :) thanks a bunch for all the hard work!!

SandervHemert, 2022-01-03:
> This issue should be now resolved in one way with the latest LineageOS 19 build (https://konstakang.com/devices/rpi4/LineageOS19/). Please test and confirm that this is the case.
> 
> I'll also include similar fix to LineageOS 18 builds if/when there's another release.

Hi KonstaT,

Thanks for this amazing build. Would this only solve it for the RPI 4 and not for the 3 like I have? 

Thanks in advance!

KonstaT, 2022-01-04:
> Would this only solve it for the RPI 4 and not for the 3 like I have?

I have retired my Pi 3 from development so I won't be providing any builds with this fix.

SandervHemert, 2022-01-04:
I get that! Thanks for getting back so quickly.

KonstaT, 2022-01-21:
This issue should be now resolved with the latest LineageOS 18 builds as well. Please test and confirm that this is the case.

KonstaT, 2022-01-22:
This has been reported to no longer be an issue with all the builds that are still maintained so closing this.

closed 2022-01-22 by KonstaT

AkshitSingh28, 2022-02-06:
I have also tried installing lineage (android 11) and (android 12) I have the same problem that the raspberry pi 4(8gb ram) is rebooting again and again

KonstaT, 2022-02-06:
> I have also tried installing lineage (android 11) and (android 12) I have the same problem that the raspberry pi 4(8gb ram) is rebooting again and again

Most likely it's not this issue if you're using the latest builds (as this was already fixed).

Attach appropriate logs (logcat, dmesg) to you issue report because it's impossible to say anything otherwise.

General advice is to use proper sdcard (at least class 10) and power source (3A in case of Pi 4).

AkshitSingh28, 2022-02-07:
I am using 4amp power supply and class 10 sd card the problem is that i have installed android 11 on raspberry pi 4 when the os stats after 5 minutes it starts rebooting and after 5 minutes again it again starts rebooting and it goes on i then tried installing android 12 but same thing happens with it

AkshitSingh28, 2022-02-07:
in the above discussion you told to diable wifi but dont know from where to diable it

KonstaT, 2022-02-07:
@AkshitSingh28 

It is not this issue so why would disabling wifi do any good? (And for how to do this see my first reply in this discussion.)

Please get appropriate log and open a new issue as this is off topic for this discussion.

[9 more replies]