lineage-rpi/android_kernel_brcm_rpi #9: Can integrate ssh and vnc

tab-1, 2020-11-17:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

KonstaT, 2020-11-17:
My LineageOS 17.1 builds already have a built-in SSH server. It requires some steps to use it because it uses key based authentication. If/when there is another release, there will be a settings option to start/stop the SSH server.

I've been told various VNC apps work so why would it need to be integrated into the build?

Graphics drivers used for hw accelerated graphics in this build rely on the EDID information your connected HDMI display sends. Please read the FAQ what you need to do if you want to run without any display connected.

tab-1, 2020-11-18:
I want to run without any display connected, but I didn't find the FAQ you mentioned, can you write me instructions?

KonstaT, 2020-11-18:
https://konstakang.com/devices/rpi4/LineageOS17.1/

If you want to run headless one option is to buy a dummy HDMI plug if you don't want to have an actual display connected.

closed 2020-11-18 by KonstaT