lineage-rpi/android_kernel_brcm_rpi #10: kernel build steps

lucastanure, 2020-12-05:
Hi, 

Can you post instructions on how to build this kernel?
What's the cross compiler? Config used?

Thanks
Lucas

KonstaT, 2020-12-05:
There's two different ways to compile the kernel. I've provided the instructions here: https://github.com/lineage-rpi/android_local_manifest/issues/8#issuecomment-625031841

For LineageOS 17.0 use lineage-17.1 branch instead of lineage-16.0.

AmbientTech, 2020-12-17:
I created a android folder in my Documents where I store all 3 git files. 

I have gotten all the way to 
ARCH=arm CROSS_COMPILE=path/to/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel- make zImage dtbs -jX

and I get this 
make: *** No rule to make target 'zImage'.  Stop.

What did I do wrong?

KonstaT, 2020-12-17:
You need to be on the root of the kernel source directory to execute
```
ARCH=arm make lineageos_rpi4_defconfig
ARCH=arm CROSS_COMPILE=path/to/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel- make zImage dtbs -jX
```

You have a used a relative path for the toolchain (home/...) where you would've needed an absolute one (/home/...).

ezebongiovi, 2020-12-28:
Hi @KonstaT

Thank you for all your job on this, I really appreciate your commitment with the Open Source community. I'm sorry to bother you but I've tried your steps and it works just great. Probably my lack of knowledge got me stuck on the kernel build step.

After executing `mka kernel` the kernel gets built correctly. Unfortunately for me I don't know how to move forward in order to get the OS installed on the Raspberry PI 4.

I've tried executing `make ramdisk systemimage vendorimage` as indicated [Here](https://github.com/android-rpi/device_arpi_rpi4) but the build failed with this error:

[image]

Did I move forward in a wrong direction or is there any problem with the build?

ezebongiovi, 2020-12-29:
Hi @KonstaT I'm having this error when trying to build lineage-16.0
[image]

KonstaT, 2020-12-29:
@ezebongiovi As commented in https://github.com/lineage-rpi/android_local_manifest/issues/8#issuecomment-625031841 I haven't released the full device specific source code for my LineageOS 16.0 & 17.1 builds (both Pi 3 & Pi 4). You can use these steps to build Linux kernel and other GPL licensed parts included in my images. There's also source code for the graphics drivers used in my builds if someone wants to work on that.

wilsavery1, 2021-03-25:
Hi KonstaT,
 I follow the step by step to Building the kernel as standalone.
for some reason I keep getting this errors.
root@builder1:**~/kernel/android_kernel_brcm_rpi# ARCH=arm make lineageos_rpi4_defconfig**

WARNING: unmet direct dependencies detected for GENERIC_GETTIMEOFDAY
  Depends on [n]: HAVE_GENERIC_VDSO [=n]
  Selected by [y]:
  - HAVE_ARM_ARCH_TIMER [=y] && CPU_V7 [=y]

WARNING: unmet direct dependencies detected for GENERIC_GETTIMEOFDAY
  Depends on [n]: HAVE_GENERIC_VDSO [=n]
  Selected by [y]:
  - HAVE_ARM_ARCH_TIMER [=y] && CPU_V7 [=y]
#
# configuration written to .config

**ARCH=arm CROSS_COMPILE=/root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel- make zImage dtbs -j4**
ake: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: No such file or directory
scripts/kconfig/conf  --syncconfig Kconfig
./scripts/gcc-version.sh: 17: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 18: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 19: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 17: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 18: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 19: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/clang-version.sh: 11: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
init/Kconfig:34:warning: 'GCC_VERSION': number is invalid

Maybe you can point me in the right direction.

Thank you.

KonstaT, 2021-03-26:
@wilsavery1 Check the path for the toolchain.

Where did you clone it? Did you clone the right branch (pie-release)? Change into the directory you cloned it and run 'pwd' to get the path.

wilsavery1, 2021-03-26:
Thank you @KonstaT for you prompt response.
All source I got following the above instruction:
Clone kernel source (https://github.com/lineage-rpi/android_kernel_brcm_rpi/tree/lineage-17.1).
Clone proprietaries to get firmware files (https://github.com/lineage-rpi/proprietary_vendor_brcm/tree/lineage-17.1).
Clone AOSP toolchain (https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/+/refs/heads/pie-release).

Now I encounter another issue (bellow)
Please note that my building machine is a MacOS.

Thank you again for your help.

scripts/extract-cert.c:21:10: scripts/sortextable.cfatal error:: 23:10: 'openssl/bio.h'fatal error : file not found'elf.h' 
file not found
#include <elf.h>#include <openssl/bio.h>

         ^~~~~~~         ^~~~~~~~~~~~~~~

scripts/recordmcount.c:28:10: fatal error: 'elf.h' file not found
#include <elf.h>
         ^~~~~~~
1 error generated.
make[2]: *** [scripts/extract-cert] Error 1
make[2]: *** Waiting for unfinished jobs....
  DTCO    arch/arm/boot/dts/overlays/act-led.dtbo
  DTC     arch/arm/boot/dts/bcm2708-rpi-b-plus.dtb
In file included from scripts/selinux/mdp/mdp.c:36:
In file included from ./security/selinux/include/classmap.h:2:
In file included from ./include/uapi/linux/capability.h:17:
./include/uapi/linux/types.h:5:10: fatal error: 'asm/types.h' file not found
#include <asm/types.h>
         ^~~~~~~~~~~~~
In file included from scripts/selinux/genheaders/genheaders.c:18:
In file included from ./security/selinux/include/classmap.h:2:
In file included from ./include/uapi/linux/capability.h:17:
./include/uapi/linux/types.h:5:10: fatal error: 'asm/types.h' file not found
#include <asm/types.h>
         ^~~~~~~~~~~~~

KonstaT, 2021-03-26:
Above instructions also specify that you clone the correct branches.

```
git clone https://github.com/lineage-rpi/android_kernel_brcm_rpi -b lineage-17.1
git clone https://github.com/lineage-rpi/proprietary_vendor_brcm -b lineage-17.1
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9 -b pie-release
```

There's a different toolchain for Mac (assuming you're using x86 one and not the new ARM kind).

`git clone https://android.googlesource.com/platform/prebuilts/gcc/darwin-x86/arm/arm-linux-androideabi-4.9 -b pie-release`

wilsavery1, 2021-03-26:
Thank again @KonstaT.
I cloned the toolchain, double, and triple check my environment, made sure all path are correct.

I'm still getting the same error, still no luck.

scripts/sortextable.c:23:10: fatal error: **'elf.h'** file not found
**#include <elf.h>**
         ^~~~~~~
  DTC     arch/arm/boot/dts/bcm2708-rpi-b-plus.dtb
scripts/recordmcount.c:28:10: fatal error: 'elf.h' file not found
**#include <elf.h>**
         ^~~~~~~
scripts/extract-cert.c:21:10: fatal error: 'openssl/bio.h' file not found
**#include <openssl/bio.h>**
         ^~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [scripts/extract-cert] Error 1
make[2]: *** Waiting for unfinished jobs....
  DTCO    arch/arm/boot/dts/overlays/adafruit18.dtbo
1 error generated.
In file included from scripts/selinux/mdp/mdp.c:36:
In file included from ./security/selinux/include/classmap.h:2:
In file included from ./include/uapi/linux/capability.h:17:
./include/uapi/linux/types.h:5:10: fatal error: 'asm/types.h' file not found
#include <asm/types.h>
         ^~~~~~~~~~~~~
make[2]: *** [scripts/sortextable] Error 1
In file included from scripts/selinux/genheaders/genheaders.c:18:
In file included from ./security/selinux/include/classmap.h:2:
In file included from ./include/uapi/linux/capability.h:17:
./include/uapi/linux/types.h:5:10: fatal error: 'asm/types.h' file not found
#include <asm/types.h>
         ^~~~~~~~~~~~~

KonstaT, 2021-03-27:
I'm not a MacOS user so I really don't know about that.

Google searching the first error tells that it seems to be some common problem on MacOS trying to build Linux natively.

Problem with missing openssl headers can happen on Linux as well and for that you need to install libssl-dev package. No idea what is the MacOS equivalent.

EPNW, 2021-05-15:
I need andriod on my rpi3 with alsa snd loop modul active, so I tried to build the kernel.
Therefor, I'm using docker. I'll leave the dockerfile here incase anyone needs to build the kernel.
NOTE that I'm not entirly sure if its ok to mix the branches `lineage-17.1-4.19` and `lineage-17.1` as described in the dockerfile, so mabye you want to replace `lineage-17.1-4.19` with `lineage-17.1` if you are not targeting the rpi3. Oh and make sure to remove the `CONFIG_SND_ALOOP` line if you don't need the alsa snd loop modul.
```
FROM ubuntu:focal
	
# Install dependencies
# Installing is split into three parts on purpose
# Combining the first two might fail because of tzdata
# After the second part, we installed all the basics so git cloning is possible
# Then we clone before installing the third dependency part, since clonig takes time
# and dependencies in the third part may change (and if they do, we don't have to clone
# again due to build caching)
RUN apt-get update \
        && TZ="Europe/Germany" DEBIAN_FRONTENTD="noninteractive" apt-get install -y --no-install-recommends \
		tzdata
RUN apt-get update \
        &&DEBIAN_FRONTENTD="noninteractive" apt-get install -y --no-install-recommends \
        nano \
        wget \
        tmux \
		git git-core\
		curl \
		ca-certificates ca-certificates-java ca-certificates-mono

# Clone the gits, do it in three commands since target branches may change
WORKDIR /app
RUN git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9 -b pie-release
RUN git clone https://github.com/lineage-rpi/proprietary_vendor_brcm -b lineage-17.1
# I'm using the lineage-17.1-4.19 branch here instead of the lineage-17.1 branch since the later seems to lack the lineageos_rpi3_defconfig file
RUN git clone https://github.com/lineage-rpi/android_kernel_brcm_rpi -b lineage-17.1-4.19

# Install dependencies, part 3
RUN apt-get update \
        &&DEBIAN_FRONTENTD="noninteractive" apt-get install -y --no-install-recommends \
		make gcc flex bison build-essential zip unzip python3 bc \
		libssl-dev \
		openjdk-8-jdk gnupg gperf \
		zlib1g-dev gcc-multilib g++-multilib libc6-dev-i386 lib32ncurses5-dev \
		x11proto-core-dev libx11-dev lib32z-dev ccache libgl1-mesa-dev libxml2-utils \
		xsltproc liblz4-tool vboot-utils u-boot-tools \
		device-tree-compiler 
RUN update-alternatives --install /usr/bin/python python /usr/bin/python3 10

# Prepare build environment
ENV BUILD_CORES=4
ENV PI_GENERATION=4
ENV ARCH=arm
ENV CROSS_COMPILE=/app/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-
# Update firmware path
RUN sed -i 's+CONFIG_EXTRA_FIRMWARE_DIR="../../../vendor/brcm/rpi+CONFIG_EXTRA_FIRMWARE_DIR="/app/proprietary_vendor_brcm/rpi+g' /app/android_kernel_brcm_rpi/arch/arm/configs/lineageos_rpi${PI_GENERATION}_defconfig
# Enable alsa loopback
RUN sed -i 's+# CONFIG_SND_ALOOP is not set+CONFIG_SND_ALOOP=m+g' /app/android_kernel_brcm_rpi/arch/arm/configs/lineageos_rpi${PI_GENERATION}_defconfig

# Actual building
WORKDIR /app/android_kernel_brcm_rpi
RUN make lineageos_rpi${PI_GENERATION}_defconfig
RUN make zImage dtbs -j${BUILD_CORES}
# Get the result
WORKDIR /app
RUN mv /app/android_kernel_brcm_rpi/arch/arm/boot/zImage /app/kernel_rpi${PI_GENERATION}.img
```
So, now my question is, after building, how can I incorperate my kernel into the images you provide (I'm refering to [these](https://konstakang.com/devices/rpi3/LineageOS17.1/))?

KonstaT, 2021-05-16:
@EPNW Rather build the snd loop driver into the kernel instead of using a module (CONFIG_SND_ALOOP=y).

You can simply replace the kernel zImage on the boot partition of the Android image with the kernel you've built yourself.

adriano98bis, 2021-11-21:
Hello, I followed the instructions to compile a new version of the lineage os kernel for raspberry pi from my laptop to which I added a usb memory unit with the installation of the latest version of ubuntu, but typing the codes specified in the instructions the ubunto terminal cannot find the required files?

Thanks for the attention.

KonstaT, 2021-11-21:
> ... but typing the codes specified in the instructions the ubunto terminal cannot find the required files?

Which steps did you follow? Can't find what required files?

3ndym10n, 2021-11-21:
From #24. I followed your first reply in this issue and used the second method.

KonstaT, 2021-11-21:
> From #24. I followed your first reply in this issue and used the second method.

Please be more precise on the exact steps. What OS you using to build?

adriano98bis, 2021-11-21:
I performed the first step specified at the beginning of the instructions: 
repo init -u git://github.com/LineageOS/android.git -b lineage-17.1
curl --create-dirs -L -o .repo/local_manifests/manifest_brcm_rpi.xml -O -L https://raw.githubusercontent.com/lineage-rpi/android_local_manifest/lineage-17.1/manifest_brcm_rpi.xml
repo sync
but the ubuntu 18.04 terminal gives the following result:
[image]

KonstaT, 2021-11-21:
> ... but the ubuntu 18.04 terminal gives the following result:

It even tells you exactly what to do. Also see https://source.android.com/setup/develop#installing-repo

3ndym10n, 2021-11-21:
> > From #24. I followed your first reply in this issue and used the second method.
> 
> Please be more precise on the exact steps. What OS are you using to build?

I was using ubuntu 20.04 in win10 wsl2.

I cloned the latest kernel source (https://github.com/lineage-rpi/android_kernel_brcm_rpi/tree/lineage-19.0), the corresponding firmware files (https://github.com/lineage-rpi/proprietary_vendor_brcm/tree/lineage-19.0) and 64-bit AOSP toolchains.

Then I made directories "vendor/brcm/" in the root of the kernel source, and copy the "rpi4/" directory into it.

At last I ran the commands to make Image then the error appeared.
```
ARCH=arm64 CROSS_COMPILE=path/to/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel- make lineageos_rpi4_defconfig
ARCH=arm64 CROSS_COMPILE=path/to/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel- make Image dtbs -jX
```

KonstaT, 2021-11-21:
@3ndym10n

Did you actually clone the repository with the git clone command? There's a lot of symlinks in the dts/overlays that can't break.

Firmware is also in the wrong place (unless you modified the line in defconfig as instructed) but it doesn't the error you saw.

adriano98bis, 2021-11-21:
> > ... ma il terminale ubuntu 18.04 dà il seguente risultato:
> 
> Ti dice anche esattamente cosa fare. Vedi anche https://source.android.com/setup/develop#installing-repo

i ran the instructions on the page, on the ubuntu terminal but the result doesn't change

adriano98bis, 2021-11-21:
since using a terminal to find the files to download to compile them into an img file. for raspberry pi it involves many errors, it would not be easier to download the build files and modify them with a program that later converts them into an img file. ?

KonstaT, 2021-11-21:
> i ran the instructions on the page, on the ubuntu terminal but the result doesn't change
I'm sure it does. If you install repo you can also use it to download the source code. Setitngs up the build environment is just something that you need to figure out.

> since using a terminal to find the files to download to compile them into an img file. for raspberry pi it involves many errors, it would not be easier to download the build files and modify them with a program that later converts them into an img file. ?
Not sure you quite understand how this works. There's only bunch of source code that eventually gets compiled and packaged into a flashable image. But yes, if you managed to compile the kernel you would use it replace the /boot/zImage (or Image on 64-bit) on the existing Android image.

adriano98bis, 2021-11-21:
> Not sure you quite understand how this works. There's only bunch of source code that eventually gets compiled and 
packaged into a flashable image. But yes, if you managed to compile the kernel you would use it replace the /boot/zImage (or Image on 64-bit) on the existing Android image.

yes even if in this case the programs that make up the operating system will be grouped in a list and then used to make up the operating system which in turn is saved as an img file. to be installed.
However i am interested in making a simple modification to lineage, which concerns the multi-cam functions to be able to use apps with multiple usb webcams at the same time?

KonstaT, 2021-11-22:
> However i am interested in making a simple modification to lineage, which concerns the multi-cam functions to be able to use apps with multiple usb webcams at the same time?

And this would have what to do with the kernel? Android camera HAL I'm using on LineageOS 16/17/18 builds already supports up to three cameras (1 x camera module + 2 USB webcams or 3 USB webcams).

adriano98bis, 2021-11-22:
I'm trying an app that uses 3 usb webcams, but every time I select a new one the previous one is deactivated rather than continue to work, this is the code:
WebCamDevice[] devices = WebCamTexture.devices;
        foreach (WebCamDevice device in devices)
        {
            if (GUILayout.Button(device.name, GUILayout.Width(200), GUILayout.Height(40)))
            {
                _mSelected = true;
                _mTexture = new WebCamTexture(device.name, 128, 128, 30);
                _mTexture.Play();
                _mWebCamMaterial.mainTexture = _mTexture;
            }
        }
reading the articles on the android site I found that the system groups all the physical webcams into a single logical webcam, so I think a kernel modification is needed; or is it my code that prevents me from using 3 webcams?

KonstaT, 2021-11-23:
@adriano98bis 
Sorry, no idea but most likely nothing to do with the kernel.

Source code for the camera HAL is now here (https://github.com/lineage-rpi/android_hardware_camera-v4l2/tree/lineage-18.1).

KonstaT, 2022-02-26:
@NumberOneGit There is no error there (just warning).

You need to use 64-bit toolchain and defconfig for lineage-18.1 and lineage-19.0 branches.

NumberOneGit, 2022-02-27:
The main issue I was referencing was the other portion:
ARCH=arm CROSS_COMPILE=/root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel- make zImage dtbs -j4
ake: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: No such file or directory
scripts/kconfig/conf --syncconfig Kconfig
./scripts/gcc-version.sh: 17: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 18: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 19: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 17: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 18: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/gcc-version.sh: 19: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
./scripts/clang-version.sh: 11: /root/kernel/arm-linux-androideabi-4.9/bin/arm-linux-androidkernel-gcc: not found
init/Kconfig:34⚠️ 'GCC_VERSION': number is invalid

Thanks for the advice, I will try to figure that out

KonstaT, 2022-02-27:
@NumberOneGit That's not the actual error message either.

You're still trying to build 32-bit (arm) kernel.

NumberOneGit, 2022-02-27:
Ah, sorry I misunderstood your previous comment as I was explicitly following the 18.1 instructions. Thanks for the clarification!

NumberOneGit, 2022-02-27:
I get the same results with aarch64:
ARCH=arm CROSS_COMPILE=/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel- make lineageos_rpi4_defconfig
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error

NumberOneGit, 2022-02-27:
I get the same results with:
ARCH=arm64 CROSS_COMPILE=/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel- make lineageos_rpi4_defconfig

But are you saying the issue is that I've downloaded the wrong source code?

KonstaT, 2022-02-27:
I have no way of knowing what you've done (or what you've downloaded). I can't help either if you don't post the output with the actual error message.

One of your edited/removed messages implied that you've given the wrong path for the toolchain. Based on the 'pi' username you're trying to do build this on Raspberry Pi which is obviously not going to work (these are x86 toolchains used to crosscompile arm executables on x86 architecture).

NumberOneGit, 2022-02-27:
ARCH=arm64 CROSS_COMPILE=/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel- make Image dtbs -j8
arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
arch/arm64/Makefile:38: LSE atomics not supported by binutils
arch/arm64/Makefile:52: Detected assembler with broken .inst; disassembly will be unreliable
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
scripts/kconfig/conf  --syncconfig Kconfig
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
init/Kconfig:34:warning: 'GCC_VERSION': number is invalid
[... 83 lines cut ...]
make[2]: *** [scripts/Makefile.lib:319: arch/arm64/boot/dts/overlays/adau1977-adc.dtbo] Error 1
make[1]: *** [scripts/Makefile.build:553: arch/arm64/boot/dts/overlays] Error 2
  CC      scripts/mod/devicetable-offsets.s
make: *** [Makefile:1359: dtbs] Error 2
make: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
Traceback (most recent call last):
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 66, in <module>
    main(sys.argv)
[... cut ...]
make[1]: *** Waiting for unfinished jobs....
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 63, in main
    cw.invoke_compiler()
  File "/home/pi/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel-gcc", line 58, in invoke_compiler
    os.execv(self.argv0, self.execargs)
OSError: [Errno 8] Exec format error
make[1]: *** [scripts/Makefile.build:99: scripts/mod/devicetable-offsets.s] Error 1
make: *** [Makefile:1211: prepare0] Error 2

KonstaT, 2022-02-28:
@NumberOneGit Like said, you can't use Raspberry Pi to compile this (using AOSP toolchains at least). Android build is only supported linux-86 (MacOS is also no longer supported https://source.android.com/setup/build/initializing). Raspberry Pi is an arm device which is also why you need to use specific toolchain to compile on x86 architecture.

NumberOneGit, 2022-02-28:
Ah, sorry.  I seem to be replying to certain comments without having refreshed the page to see your updated common.  I think you added the second paragraph above after typing the first portion and I replied without refreshing the page.  I'll try that all now, thanks for the help!

NumberOneGit, 2022-03-03:
If the kernel compiles properly will there be a successful build message at the end?  After rolling back to an earlier Ubuntu, everything seemed to go mostly fine with a few minor errors.  Will the final product of a successful build be an Image file in the /android_kernel_brcm_rpi folder?

KonstaT, 2022-03-03:
Build products will be in kernel source's arch/arm64/boot/ directory unless you've specified some other output directory.

NumberOneGit, 2022-03-03:
It seems to get a good deal of the way through (after a regulatory.db error at the beginning), but ultimately ends with these errors.  I decided to try to compile the kernel without changes until I could get it to work as is explicitly following the directions and these were results:

ARCH=arm64 CROSS_COMPILE=/home/tzar/lineage18/aarch64-linux-android-4.9/bin/aarch64-linux-androidkernel- make Image dtbs -j8 -k
  CALL    scripts/atomic/check-atomics.sh
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/regulatory.db', needed by 'drivers/base/firmware_loader/builtin/regulatory.db.gen.o'.
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/regulatory.db.p7s', needed by 'drivers/base/firmware_loader/builtin/regulatory.db.p7s.gen.o'.
  UPD     drivers/base/firmware_loader/builtin/regulatory.db.p7s.gen.S
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/brcm/brcmfmac43455-sdio.bin', needed by 'drivers/base/firmware_loader/builtin/brcm/brcmfmac43455-sdio.bin.gen.o'.
  UPD     drivers/base/firmware_loader/builtin/brcm/brcmfmac43455-sdio.bin.gen.S
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/brcm/brcmfmac43455-sdio.clm_blob', needed by 'drivers/base/firmware_loader/builtin/brcm/brcmfmac43455-sdio.clm_blob.gen.o'.
  UPD     drivers/base/firmware_loader/builtin/brcm/brcmfmac43455-sdio.clm_blob.gen.S
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/brcm/brcmfmac43455-sdio.txt', needed by 'drivers/base/firmware_loader/builtin/brcm/brcmfmac43455-sdio.txt.gen.o'.
  UPD     drivers/base/firmware_loader/builtin/brcm/brcmfmac43455-sdio.txt.gen.S
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/brcm/brcmfmac43456-sdio.bin', needed by 'drivers/base/firmware_loader/builtin/brcm/brcmfmac43456-sdio.bin.gen.o'.
  UPD     drivers/base/firmware_loader/builtin/brcm/brcmfmac43456-sdio.bin.gen.S
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/brcm/brcmfmac43456-sdio.clm_blob', needed by 'drivers/base/firmware_loader/builtin/brcm/brcmfmac43456-sdio.clm_blob.gen.o'.
  UPD     drivers/base/firmware_loader/builtin/brcm/brcmfmac43456-sdio.clm_blob.gen.S
make[4]: *** No rule to make target '../../../vendor/brcm/rpi4/proprietary/vendor/firmware/brcm/brcmfmac43456-sdio.txt', needed by 'drivers/base/firmware_loader/builtin/brcm/brcmfmac43456-sdio.txt.gen.o'.
  UPD     drivers/base/firmware_loader/builtin/brcm/brcmfmac43456-sdio.txt.gen.S
make[4]: Target '__build' not remade because of errors.
make[3]: *** [scripts/Makefile.build:553: drivers/base/firmware_loader/builtin] Error 2
make[3]: Target '__build' not remade because of errors.
make[2]: *** [scripts/Makefile.build:553: drivers/base/firmware_loader] Error 2

KonstaT, 2022-03-03:
Please read the instructions.

> Modify firmware path according to where you've saved them (https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-18.1/arch/arm/configs/lineageos_rpi4_defconfig#L1424 for 32-bit or https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-18.1/arch/arm64/configs/lineageos_rpi4_defconfig#L1439 for 64-bit).

NumberOneGit, 2022-03-03:
Thanks!  I had missed that step this time around.  Finally got it working.  I appreciate your help!