lineage-rpi/android_kernel_brcm_rpi #11: Goodix i2c touchscreen

goutx2, 2021-01-15:
First of all thanks for this great work I have a rpi4 4 gb and 7 inche i2c goodix touchscreen 
For the official raspberry os i just add a line to config.txt file to make it work 
dtoverlay=goodix
And i2c=on 
But it not work on lenageos
I tried i2cdetect  -1 
And i can see 14 reg so the touchscreen work but how to load the driver
What i have to do to make work

KonstaT, 2021-01-16:
Can you provide a bit more detail about the touchscreen. Does it have HDMI interface for display and I2C is used for touch input?

Goodix touchscreen driver is not built into the kernel so that is why it's not loaded.
https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-17.1/arch/arm/configs/lineageos_rpi4_defconfig#L2273

You would need to recompile the kernel with that option enabled. There build instructions for the kernel available here:
https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/10#issuecomment-739346981

You might need to create an IDC file for it on the Android side as well (https://source.android.com/devices/input/input-device-configuration-files & e.g. /vendor/usr/idc/ADS7846_Touchscreen.idc in my builds for reference).

goutx2, 2021-01-16:
Thank you for your reply. 
So it's two sperates
[image]
[image]

 parts i connect display to HDMI
And touchscreen to i2c interface its kind of homebrew work 
From what i understand i have to recompile the kernel than flash it via twrp ?

KonstaT, 2021-01-16:
In that case I think you should be able to get this working.

When you compile the kernel you'll get zImage that you can simply copy to the boot partition to replace the original kernel (/boot/zImage).

Check 'dmesg' output on booting the device that the driver is loaded. You should also see the it listed as input device (see /sys/class/input*).

goutx2, 2021-01-16:
Thanks i will close the issue

KonstaT, 2021-01-16:
Did you get this working? What steps did it require?

I could include the necessary changes in future builds to make it easier for the next person with goodix touchscreen.

goutx2, 2021-01-16:
Not yet  i have to recompile the kernel as you said but to be honest i am not familiar with this staff
I will appreciate if you could make zimage for me and i will report if it will work

closed 2021-01-17 by KonstaT

goutx2, 2021-01-23:
> Can you provide a bit more detail about the touchscreen. Does it have HDMI interface for display and I2C is used for touch input?
> 
> Goodix touchscreen driver is not built into the kernel so that is why it's not loaded.
> https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-17.1/arch/arm/configs/lineageos_rpi4_defconfig#L2273
> 
> You would need to recompile the kernel with that option enabled. There's build instructions for the kernel available here:
> [#10 (comment)](https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/10#issuecomment-739346981)
> 
> You might need to create an IDC file for it on the Android side as well (https://source.android.com/devices/input/input-device-configuration-files & see e.g. /vendor/usr/idc/ADS7846_Touchscreen.idc in my builds for reference).

After a week finally i get zImage i recompile d the kernel with goodix touchscreen driver built in and this is the first step . just tell me how to add IDC fille for goodix from what i understand i have to rename ADS7846_Touchscreen.idc to goodix.idc is this right?

goutx2, 2021-01-23:
Hi again i confirm it work perfectly with out adding any IDC file 
Thes is my steps to make it work 
Modify
 https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-17.1/arch/arm/configs/lineageos_rpi4_defconfig#L2273 
CONFIG_TOUCHSCREEN_GOODIX=y
Then compile the kernel and copy paste zImage
Modify config.txt 
dtparam=i2c_arm=on
dtoverlay=goodix 
Wiring following the picture
And booom it work
[image]

KonstaT, 2021-01-24:
Cool, good to hear you got it figured out.

I'll enable the goodix touchscreen driver for future builds.