lineage-rpi/android_kernel_brcm_rpi #12: USB Sound card

Rpi4Alex, 2021-01-27:
Hi, I decided to return to the issue with sound card attached to usb. Here is dmesg and logcat. In fact it looks like it is determined by system but for some reasons it doesnt work (neither mic nor earphones)

logcat
1970-01-01 03:03:04.687 0-0/? I/usb 1-1.3: Product: C-Media USB Headphone Set
1970-01-01 03:03:04.771 0-0/? I/input: C-Media USB Headphone Set   as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.3/0003:0D8C:000C.0002/input/input1
1970-01-01 03:03:04.788 0-0/? I/hid-generic 0003: 0D8C:000C.0002: input,hidraw1: USB HID v1.00 Device [C-Media USB Headphone Set  ] on usb-0000:01:00.0-1.3/input3
2021-01-27 20:46:09.981 385-535/system_process D/UsbHostManager: USB device attached: vidpid 0d8c:000c mfg/product/ver/serial null/C-Media USB Headphone Set  /1.00/null hasAudio/HID/Storage: true/true/false
2021-01-27 20:46:09.986 385-535/system_process D/UsbHostManager: Added device UsbDevice[mName=/dev/bus/usb/001/004,mVendorId=3468,mProductId=12,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=null,mProductName=C-Media USB Headphone Set  ,mVersion=1.00,mSerialNumberReader=com.android.server.usb.UsbSerialReader@51f5e63, mHasAudioPlayback=true, mHasAudioCapture=true, mHasMidi=false, mHasVideoCapture=false, mHasVideoPlayback=false, mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=160,mMaxPower=50,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=1,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=1,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=9,mMaxPacketSize=200,mInterval=1]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=2,mAlternateSetting=1,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=5,mMaxPacketSize=100,mInterval=1]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=null,mClass=3,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=3,mMaxPacketSize=4,mInterval=32]]]]
2021-01-27 20:46:10.032 385-449/system_process D/EventHub: No input device configuration file found for device 'C-Media USB Headphone Set  '.
2021-01-27 20:46:10.039 385-449/system_process I/EventHub: New device: id=2, fd=219, path='/dev/input/event1', name='C-Media USB Headphone Set  ', classes=0x80000001, configuration='', keyLayout='/vendor/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, 
2021-01-27 20:46:10.039 385-449/system_process I/InputReader: Device added: id=3, eventHubId=2, name='C-Media USB Headphone Set  ', descriptor='8fc40c86e5ff2b922fa4093b9af21ddfba9f56c9',sources=0x00000101
2021-01-27 20:46:10.049 1534-1534/com.stevesoltys.seedvault D/UsbIntentReceiver:   name: null C-Media USB Headphone Set  
2021-01-27 20:46:10.317 1072-1656/org.lineageos.audiofx D/AudioFx-AudioOutputChangeListener: onAudioOutputChanged id: 10 type: 22 name: USB-Audio - C-Media USB Headphone Set address: card=2;device=0; [android.media.AudioDeviceInfo@29]
...
extract for alsa

1970-01-01 03:00:17.644 383-383/? E/UsbAlsaJackDetectorJNI: Can't register UsbAlsaJackDetector native methods
2021-01-27 21:33:31.367 383-537/system_process I/UsbAlsaDevice: OUTPUT JACK connected: true
2021-01-27 21:33:31.370 383-537/system_process I/UsbAlsaDevice: INPUT JACK connected: true
2021-01-27 21:37:01.077 383-537/system_process I/UsbAlsaDevice: OUTPUT JACK connected: true
2021-01-27 21:37:01.081 383-537/system_process I/UsbAlsaDevice: INPUT JACK connected: true

dmesg
[  105.402743] type=1400 audit(1611769490.563:308): avc: denied { syslog_read } for comm="dmesg" scon
[  152.055097] type=1400 audit(1611769537.213:309): avc: denied { read } for comm="drm-event-liste" p
hal_graphics_composer_default:s0 tclass=netlink_kobject_uevent_socket permissive=1
[  153.060465] nf_conntrack: default automatic helper assignment has been turned off for security rea
[  184.536670] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[  184.671146] usb 1-1.3: New USB device found, idVendor=0d8c, idProduct=000c, bcdDevice= 1.00
[  184.679678] usb 1-1.3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  184.687160] usb 1-1.3: Product: C-Media USB Headphone Set
[  184.771008] input: C-Media USB Headphone Set   as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.3/0003:0D8C:000C.0003/input/input2
[  184.788570] hid-generic 0003:0D8C:000C.0002: input,hidraw1: USB HID v1.00 Device [C-Media USB Headphone Set  ] on usb-0000:01:00.0-1.3/input3

KonstaT, 2021-01-28:
Please specify system version and device. LineageOS 16, 17, 18?

I don't see anything wrong with that. Also my USB soundcard works just fine on all versions. Are you sure the issue just isn't the volume level? Go to audio settings and increase the volume to the max.

Rpi4Alex, 2021-01-28:
Thanks.
It is all about 18 (but situation is the same for LOs17). 
I'll check but waht about mic defenitely ot is not a volum issue?

KonstaT, 2021-01-28:
AFAICT the volume level affects both input & output.

Rpi4Alex, 2021-01-28:
THanks again.
Earphones are done - they work (the reason was indeed vlume level). Is it possible to save it at a required level once and forever. Becuse it goes to initial level upon reboot so it is required to return back to settings and set it each and every time manually.  
Mic doesnt want to work... :(  
Any thoughts?

KonstaT, 2021-01-28:
Have you tested the mic with the built-in recorder application? There been some reports that mic is not working on some third party apps on LineageOS 18.

Rpi4Alex, 2021-01-28:
thanks. Of cource it is internal recorder because for the clarity of experiment nothing else is installed.

Rpi4Alex, 2021-01-30:
Hi, thank you once again for the support and advice. It looks like the topic can be closed since i found the solution. I've changed the mic and now it works fine. Although i still havent got an understanding on why the same mic worked perfectly well on the same USB sound card with LoS 16.

closed 2021-01-30 by KonstaT