lineage-rpi/android_kernel_brcm_rpi #14: question: when we will get 64 bit android builds? (arm64-v8a)

Id-ware, 2021-02-27:
(i couldn't find a place to ask it directly, so i post here apologies if it's wrong.)

hi, 

after installing lineage os for the rpi, i noticed it's only support 32 bit android apps (armeabi-v7a)
while the rpi 4 cpu have a support for 64 bit architecture 
(it can support arm64-v8a, for example OMNI android for the rpi 4 have 64 bit builds)

so i wonder, why lineage-os only have 32 bit builds?
and will we see 64 bit support in the future?

thanks!

Id-ware, 2021-02-27:
nevermind, i found already an answer:
https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/2#issue-630183203

closed 2021-02-27 by Id-ware

KonstaT, 2021-02-28:
Here a bit more thorough answer that isn't an year old.
https://forum.xda-developers.com/t/dev-rom-unofficial-lineageos-18-1-android-11-for-raspberry-pi-4-b.4212945/post-84352695