lineage-rpi/android_kernel_brcm_rpi #17: Add new soundcard

hackintoshlover, 2021-06-17:
Thanks for the amazing project.
I am running lineage is 18.1 on a raspberry pi 4 4gb variant.
I have a wm8960 soundcard from waveshare.
Can you tell me how I can add and activate the same in the current version?

KonstaT, 2021-06-18:
According to the documentation (https://www.waveshare.com/wiki/WM8960_Audio_HAT) you need to add a kernel driver for this hardware (https://github.com/waveshare/WM8960-Audio-HAT).

It's not possible to build the kernel on the device on Android so you'd need to find a way to add the driver. Here's build instructions for the Android kernel: https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/10#issuecomment-739346981

hackintoshlover, 2021-06-19:
The driver for wm8960 is already present in the raspberry pi kernel 5.10
So I need help for two things
How do I compile the new kernel for lineage rpi?
How do I enable it in the lineage Android sound settings menu?

KonstaT, 2021-06-19:
I'm not planning on switching to 5.10 kernel until Android 12.

You would need bring up the 5.10 kernel for Android by merging Raspberry Pi kernel and Android kernel from AOSP. Source code in this repository can be used as reference how it was done & configured for 5.4. It's not something that I give you any step by step guide.

hackintoshlover, 2021-06-19:
Nevermind
Thanks for no help

closed 2021-06-19 by hackintoshlover

KonstaT, 2021-06-19:
LOL, ok. So you weren't even looking for any help. You just wanted me to do something for you.

hackintoshlover, 2021-06-19:
The no went by mistakely
I am following all the steps you mentioned
If run into something
I'll raise a new issue