lineage-rpi/android_kernel_brcm_rpi #21: Two displays crash it...

pnill, 2021-11-02:
**Describe the bug**
Connecting a second HDMI cable causes the OS to crash (In the TV build haven't tested the non TV build yet). I tried booting with the second display connected and got the same result.

Glodroid was able to support this but the latest build pre-release doesn't seem to have the correct 3D drivers considering things were laggy.

**To reproduce**
Simply connect a second display.

**Expected behaviour**
Android to detect/extend the display to the second display and see it as a "presentation" display.

**Actual behaviour**
The system crashed continuously, or at least froze up.

**System**
There's no raspinfo on this android build.

* Which model of Raspberry Pi? RPI4 8gb
* Which OS and version (`cat /etc/rpi-issue`)? N/A your TV build
* Which firmware version (`vcgencmd version`)? N/A
* Which kernel version (`uname -a`)? Linux localhost 5.4.155-v8-g82b4c40efffe #1 SMP PREEMPT Tue Oct 26 11:17:43 EEST 2021 aarch64

**Logs**
*dmesg*
`[  443.868944] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  454.108944] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  464.348940] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CONNECTOR:38:HDMI-A-2] flip_done timed out
[  474.588924] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  484.828903] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  495.068915] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
[  505.308921] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  515.548929] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  525.788885] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
[  536.028910] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:80:plane-5] flip_done timed out
[  546.268924] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  556.508934] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  566.748922] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
[  576.988935] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:80:plane-5] flip_done timed out
[  587.228919] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  597.468921] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  607.708927] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
[  617.948938] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:80:plane-5] flip_done timed out
[  628.188935] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  638.428954] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  648.668922] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
[  658.908917] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:80:plane-5] flip_done timed out
[  669.148929] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out`

*logcat*
`
11-02 02:01:54.964   227   295 I hwc-drm-connector: force mode to 1920x1080@0Hz
11-02 02:01:54.991   227   295 I hwc-drm-connector: force mode to 1920x1080@0Hz
11-02 02:01:55.017   227   295 D hwc-drm-connector: add new mode 1920x1080@60.0 id 8 for display 1
11-02 02:01:55.017   227   295 D hwc-drm-connector: add new mode 1920x1080@75.0 id 9 for display 1
11-02 02:01:55.017   227   295 D hwc-drm-connector: add new mode 1920x1080@60.0 id 10 for display 1
11-02 02:01:55.017   227   295 D hwc-drm-connector: add new mode 1920x1080@59.9 id 11 for display 1
11-02 02:01:55.017   227   295 D hwc-drm-connector: add new mode 1920x1080@50.0 id 12 for display 1
11-02 02:01:55.018   227   295 I hwc-drm-two: Plug event @8763981346 for connector 38 on display 1
11-02 02:01:55.018   227   295 I hwc-drm-connector: force mode to 1920x1080@0Hz
11-02 02:01:55.070   243   243 E HWComposer: isConnected failed for display 1: Invalid display
11-02 02:01:55.039   227   227 I drm-event-liste: type=1400 audit(0.0:255): avc: denied { read } for path="socket:[13110]" dev="sockfs" ino=13110 scontext=u:r:hal_graphics_composer_default:s0 tcontext=u:r:hal_graphics_composer_default:s0 tclass=netlink_kobject_uevent_socket permissive=1
11-02 02:01:55.071   227   304 I hwc-drm-connector: force mode to 1920x1080@0Hz
11-02 02:01:55.099   243   243 E HWComposer: getDisplayConnectionType: getDisplayConnectionType failed for display 1: Unsupported (8)
11-02 02:01:55.099   243   243 I FramebufferSurface: framebuffer size has been limited to [1920x1080] from [1920x1080]
11-02 02:01:55.102   243   434 W SurfaceFlinger: Can't find offset for 74.97 fps
11-02 02:01:55.104   243   434 E HWComposer: getSupportedContentTypes: getSupportedContentTypes failed for display 1: Unsupported (8)
11-02 02:00:36.603   396   414 D AutofillManagerService: onBackKeyPressed()
11-02 02:01:55.105   396   416 I DisplayManagerService: Display device added: DisplayDeviceInfo{"HDMI Screen": uniqueId="local:1", 1920 x 1080, modeId 7, defaultModeId 7, supportedModes [{id=4, width=1920, height=1080, fps=50.0}, {id=5, width=1920, height=1080, fps=59.9402}, {id=6, width=1920, height=1080, fps=74.9725}, {id=7, width=1920, height=1080, fps=60.000004}], colorMode 0, supportedColorModes [0], HdrCapabilities HdrCapabilities{mSupportedHdrTypes=[], mMaxLuminance=500.0, mMaxAverageLuminance=500.0, mMinLuminance=0.0}, allmSupported false, gameContentTypeSupported false, density 213, 320.0 x 320.0 dpi, appVsyncOff 1000000, presDeadline 16666666, touch EXTERNAL, rotation 0, type EXTERNAL, address {port=1}, deviceProductInfo null, state UNKNOWN, FLAG_SECURE, FLAG_SUPPORTS_PROTECTED_BUFFERS, FLAG_PRESENTATION}
11-02 02:01:55.120   243   243 D SurfaceFlinger: Setting power mode 2 on display 1
11-02 02:01:55.121   243   243 D SurfaceFlinger: Finished setting power mode 2 on display 1
[... 53 lines cut ...]
12-31 16:13:22.268     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
12-31 16:13:32.508     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
12-31 16:13:42.748     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:80:plane-5] flip_done timed out
12-31 16:13:52.988     0     0 E         : [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
12-31 16:14:03.228     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
12-31 16:14:13.468     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
12-31 16:14:23.708     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:80:plane-5] flip_done timed out
12-31 16:14:33.948     0     0 E         : [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
12-31 16:14:44.188     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
12-31 16:14:54.428     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out
12-31 16:15:04.668     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:80:plane-5] flip_done timed out
12-31 16:15:14.908     0     0 E         : [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
12-31 16:15:25.148     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] flip_done timed out
11-02 02:05:58.247  1632  1632 I dmesg   : type=1400 audit(0.0:256): avc: denied { syslog_read } for scontext=u:r:shell:s0 tcontext=u:r:kernel:s0 tclass=system permissive=1
11-02 02:06:15.087   396   446 D WifiNl80211Manager: Scan result ready event
11-02 02:06:15.087   396   446 D WifiNative: Scan result ready event
11-02 02:06:15.089   396   413 I EthernetTracker: interfaceLinkStateChanged, iface: wlan0, up: true
[... cut ...]
12-31 16:15:35.388     0     0 E         : [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] flip_done timed out`

**Additional context**
Both displays are 1080p 60hz, connected via the micro HDMI ports both work independently on their own, and this works in glodroid as well as standard android just not this build for some reason; was gonna try enabling the CRTC stuff in the raspi settings since it seems to complain about that but really have no idea what's going on haven't dug far enough into the logs above to determine why it's failing.

KonstaT, 2021-11-02:
Thanks for the detailed report.

Dual HDMI is not well supported (if at all) with the KMS driver on Linux 5.4 kernel. Pi 4 KMS driver was further improved in 5.10 kernel where this should be possible among other things (DSI display support, etc). Plan was to switch to 5.10 with Android 12 but that's probably now also on how due to other issues (no HDMI audio available on that one).

There's also no support for switching between display on Android side so there's a long way to go for anything like this to be actually usable.

pnill, 2021-11-02:
So, I don't know much about how you built this and hadn't dug in so I may ask some potentially idiotic questions but start with I appreciate everything you've done with this I'm sure other people are getting a lot of use out of it and it's cool to see android running on Pi especially a better performing version than the glodroid pre-release that's 'released'.

Does glodroid not use the KMS driver? or the 5.4 kernel? 
Would it be possible for me to build the 5.10 kernel and kind of mix and match with your android 11 build?

Last, I don't think switching between displays is really necessary. If you mean like user-based switching, as far as I understand one display becomes the 'primary' display and the other is the 'presentation' display sort of like how casting makes an additional virtual display in the context of the Android API from the software/userland prospective.

KonstaT, 2021-11-02:
Roman from GloDroid has done the work upstreaming the required graphics support so it's mostly thanks to him how things in that regard are set up in my builds as well. There shouldn't be any major difference in performance unless you're using some very high display resolution (4k) on other.

Yes, GloDroid uses KMS but 5.10 kernel.

pnill, 2021-11-02:
> Yes, GloDroid uses KMS but 5.10 kernel.

Is that graphics support/5.10 kernel there in the v6.0 pre-release or is that stuff from his "Next" branch? I was running the 6.0 pre-release image; still trying to get the repo to sync in order to build the Next branch.

But the performance was terrible even at the launch screen both displays were 1080p, meanwhile running this everything ran great minus that display support so it was odd. I had seen you did borrow some code from there and figured it may perform just the same but was surprised when it performed really well.

KonstaT, 2021-11-02:
I'm not the right person to ask and this is not the right place to discuss it.

closed 2021-11-02 by KonstaT