lineage-rpi/android_kernel_brcm_rpi #25: Full build but not get the zImage.img

ilike28, 2021-12-05:
Dear Sirs

I used the command which is "mka -j4 kernel ramdisk systemimage vendorimage".
But It doesn't produce the boot image.
Could you please tell me how can make it @@?

Thanks ^_^

[image]

KonstaT, 2021-12-05:
Kernel builds steps are available at https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/10#issuecomment-739346981.

The 'kernel' you've highlighted is a copy of the kernel Image (64-bit) or zImage (32-bit). Just rename the file. You can also find all the kernel build artifacts inside obj/KERNEL_OBJ/ directory on the out dir you're already at.

There's no boot image on the Pi like on regular Android devices. Boot partition is used instead.

closed 2021-12-05 by KonstaT

ilike28, 2021-12-05:
Thanjks ^_^