lineage-rpi/android_kernel_brcm_rpi #28: [Question] about porting new sensor driver

ilike28, 2021-12-16:
Can you teach me how to modify the HAL config(or code) so that HAL can recognize the newly added sensor driver?
For example : 
If I add a sensor driver under the kernel/iio folder which name is called bmc150_magn_i2c.
How can I made HAL recognize the newly added sensor ??

Thanks ^_^

[image]

KonstaT, 2021-12-16:
You shouldn't need any sensor HAL modifications for this. If the sensor has the IIO kernel driver and the sensor is of supported type it should likely just work. You need to enable the kernel driver for your sensor and a device tree overlay to enable driver/configure I2C. This only requires compiling the kernel & dtbs.

Probably can't explain it any better than I did here https://forums.raspberrypi.com/viewtopic.php?t=297349&start=125#p1927487 (read the rest of the page as well).

ilike28, 2021-12-28:
> You shouldn't need any sensor HAL modifications for this. If the sensor has an IIO kernel driver and type of the sensor is supported by the sensor HAL it should likely just work. You need to enable the kernel driver for your sensor hardware and add a device tree overlay to enable the driver/configure I2C. This only requires compiling the kernel & dtbs.
> 
> Probably can't explain it any better than I did here https://forums.raspberrypi.com/viewtopic.php?t=297349&start=125#p1927487 (read the rest of that page as well).

Could you please tell me the following question~~??

Is this IIO HAL code written by you?? Or is it originally included in the Android source code??
If I want to write HAL code by myself, do you have any recommended reference materials?

Thanks^_^

KonstaT, 2021-12-28:
> Is this IIO HAL code written by you?? Or is it originally included in the Android source code?? If I want to write HAL code by myself, do you have any recommended reference materials?

No, as you can see from the repository name, where it was forked from, and copyright headers it was initially written by Intel. No, it is not part of Google's Android source code (AOSP) but Intel's fork to support Android on their platform. One reference for IIO sensor HAL is of course the one I'm using (https://github.com/lineage-rpi/android_hardware_intel_sensors-iio) and I also have one forked from STMicro (https://github.com/lineage-rpi/android_hardware_st_sensors-iio).

> For example bmc150 is a 6-axis digital compass. How does the compass API of the Android framework know which driver to connect to?

That's exactly what the sensor HAL is for. It's between the kernel driver and the Android sensor framework (see https://source.android.com/devices/sensors/sensor-stack).

ilike28, 2021-12-29:
> > Is this IIO HAL code written by you?? Or is it originally included in the Android source code?? If I want to write HAL code by myself, do you have any recommended reference materials?
> 
> No, as you can see from the repository name, where it was forked from, and the copyright headers it was initially written by Intel. No, it is not part of Google's Android source code (AOSP) but Intel's fork to support Android on their platform (but AOSP has another IIO HAL from Intel https://android.googlesource.com/platform/hardware/intel/sensors/). One reference for IIO sensor HAL is of course the one I'm using (https://github.com/lineage-rpi/android_hardware_intel_sensors-iio) and I also have one forked from STMicro (https://github.com/lineage-rpi/android_hardware_st_sensors-iio). Android-x86 project has one additional IIO sensor HAL as well (http://git.osdn.net/view?p=android-x86/hardware-libsensors.git;a=summary).
> 
> > For example bmc150 is a 6-axis digital compass. How does the compass API of the Android framework know which driver to connect to?
> 
> That's exactly what the sensor HAL is for. It sits between the kernel driver for the sensor hardware and the Android sensor framework (see https://source.android.com/devices/sensors/sensor-stack).

The Following web page in "android_hardware_intel_sensors-iio" is not found.
Do you know where can find reference materials on these two web pages?

https://source.android.com/devices/sensors/hal-interface.html
 [basic tour of the Android sensors HAL interface]

http://source.android.com/devices/halref/sensors_8h_source.html
 [Android sensor details]

In addition, in addition to HAL and DTBO config these two blocks need to be set
Do init.rc and ueventd.rc usually need to be modified?

Thanks ^_^

KonstaT, 2022-01-01:
> Do you know where can find reference materials on these two web pages?

Android documentation should have the information that is still relevant today and for current Android versions.

> In addition, in addition to HAL and DTBO config these two blocks need to be set Do init.rc and ueventd.rc usually need to be modified?

Usually yes, depends on the sensor HAL. All the necessary permissions for the IIO sensor HAL I'm using are already there on my builds so if you're adding a sensor no modification should be necessary.