lineage-rpi/android_kernel_brcm_rpi #30: tigervnc screen didnt display mouse icon

crazyn2, 2022-11-13:
### Describe the bug

tigervnc screen didnt display mouse icon

### Steps to reproduce the behaviour

tigervnc connected vnc server of lineageos20

### Device (s)

Raspberry Pi 4 Mod. B

### System

rpi4:/ $ uname -a
Linux localhost 5.15.74-v8-ga9455b88085b #1 SMP PREEMPT Tue Oct 18 17:10:34 UTC 2022 aarch64 Toybox
rpi4:/ $ cat /etc/rpi-issue
cat: /etc/rpi-issue: No such file or directory
1|rpi4:/ $ vcgencmd version
VCHI initialization failed

### Logs

[test.log](https://github.com/lineage-rpi/android_kernel_brcm_rpi/files/9997080/test.log)

KonstaT, 2022-11-13:
Are you using the built-in VNC server (Settings -> System -> Raspberry Pi settings -> VNC)?

It works fine with Remmina that comes bundled with Ubuntu at least. I haven't tested other VNC clients. If you're running some third party VNC server on your Pi it probably doesn't have access to /dev/uinput (see e.g. http://disq.us/p/2qt1fxm).

crazyn2, 2022-11-13:
yes, im using native vnc server of pi. ill try remmina next timw

crazyn2, 2022-11-14:
remmina worked ideally. thanks for your help

closed 2022-11-14 by crazyn2