lineage-rpi/android_kernel_brcm_rpi #31: Max17040 can't enable with config_user.txt

xrh0905, 2023-04-08:
### Describe the bug

dtoverlay=i2c-sensor,max17040
Doesn't work.
System doesn't register the i2c device.

### Steps to reproduce the behaviour

```
dtoverlay=i2c-sensor,max17040
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

LineageOS 20

### Logs

```
rpi4:/ # i2cdetect 1
Probe chips 0x03-0x77 on bus 1? (Y/n):Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: 20 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- 36 -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
```
rpi4:/ # ls /sys/class/i2c-dev/i2c-1/device/
1-0068  delete_device  device  i2c-dev  name  new_device  of_node  power  subsystem  uevent
```

KonstaT, 2023-04-08:
You would need to compile the kernel with the max17040 driver enabled at least.
https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-20.0/arch/arm64/configs/lineageos_rpi4_defconfig#L2944

What hardware uses this and what is the use case for this in Android? AFAICT this is battery monitor driver.

xrh0905, 2023-04-08:
> You would need to compile the kernel with the max17040 driver enabled at least. https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-20.0/arch/arm64/configs/lineageos_rpi4_defconfig#L2944
> 
> What hardware uses this and what is the use case for this in Android? AFAICT this is a battery monitor driver.

Exactily.
It's an battery monitor that my battery board [UPS-18650-Lite](https://github.com/linshuqin329/UPS-18650-Lite) used.
It works well in Raspberry OS with default kernel, and even better with my custom one build with 'CHARGER_GPIO', which I opened a PR https://github.com/raspberrypi/linux/pull/5421 for.
It would be nice to have that in Android OS so that system can read the battery of my device.

So Android OS is using a new kernel config instead of the rpi3 one?

KonstaT, 2023-04-08:
Android uses it's custom Linux kernel. Yes, you would need to compile your own kernel that supports the hardware. There's build instructions available here. https://github.com/lineage-rpi/android_kernel_manifest/tree/lineage-20.0

Even if you added kernel driver for the battery monitor, it wouldn't do anything useful in Android UI. You would need to implement health HAL that would read the battery status. https://source.android.com/docs/core/perf/health & https://github.com/raspberry-vanilla/android_device_brcm_rpi4/tree/android-13.0/health (current health HAL implementation used on my builds that just fakes battery stats)

xrh0905, 2023-04-08:
I understand. Thanks for helping.

closed 2023-04-08 by xrh0905