lineage-rpi/android_kernel_brcm_rpi #32: Waveshare xinch DSI touchscreen support

belliash, 2023-04-19:
### Describe the bug

Is it possible to add support for Waveshare DSI touchscreen? I got this specific one: https://www.waveshare.com/wiki/7inch_DSI_LCD_(C) which requires additional work. Unfortunately, vendor publishes only binaries for some specific Linux kernel.

On the other hand, there is an open pull request: https://github.com/raspberrypi/linux/pull/5433 which enables support for most (all?) touchscreens from Waveshare. Could you include the patch from above pull request into the LineageOS build, please?

### Steps to reproduce the behaviour

https://github.com/raspberrypi/linux/pull/5433 - please include this, as there are many touchscreens available and supporting them would be a step forward to many RPi users.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Lineage 20

KonstaT, 2023-04-21:
It will be supported in my builds if/when this lands on the upstream Raspberry Pi kernel.

Looks like this is on Linux 6.1 for now and I'm not sure if it will be backported to 5.15 (or can be with reasonable effort). So likely looking at Android 14 that I'll ship with 6.1 kernel or Android 13 with separate 6.1 kernel add-on that I already have posted for the AOSP builds (you can also use it with LineageOS 20).

belliash, 2023-04-22:
I doubt it will be backported. Would be nice if you could update AOSP kernel as well as publish 6.1 based kernel for LineageOS after mentioned PR gets merged.

belliash, 2023-04-24:
https://github.com/raspberrypi/linux/pull/5433 got merged.

belliash, 2023-05-02:
I have finally managed to give it a try and I think there is something wrong.
I power rpi on, the red and green leds on rpi are solid and green led on lcd is blinking. LineageOS 20 starts booting, green led on rpi turns of and starts blinking, green led on lcd stops blinking (this is a good sign), but after a while, red led on rpi goes out for a while and this repeats all the time.

LCD is black, so I don't see any progress or errors but as red (power) led on rpi goes out for around 1sec, I think it is stuck in some boot loop.

KonstaT, 2023-05-02:
Did you install the above zip in TWRP and modify /boot/config.txt and remove /boot/resolution.txt as suggested?

Impossible to say just by the blinking LEDs without seeing any logs. You can crab logs using serial console adapter or ADB on the USB-C port (you need to have powered USB-C hub or 5V on the GPIO and USB debugging needs to be enabled beforehand from the developer options).

belliash, 2023-05-02:
I have modified config.txt and removed resolution.txt. However, I have flashed IMG file (lineage-20.0-20230321-UNOFFICIAL-KonstaKANG-rpi4.zip) with RPi Imager 1.7.4 and I have overwritten all files in /boot partition from attached kernel zip file. I cannot use TWRP as it does not support my LCD display. I cannot do anything with a blank screen.

I will try to attach serial console with UART controller and provide some logs. ADB won't work for me as well, as I don't have touchscreen working, what means I cannot do anything in the UI (i.e. enable debugging in developer options).

KonstaT, 2023-05-02:
You can of course use HDMI display to boot to Android/TWRP first.

Using the DSI display after that would only require adding the one line to /boot/config.txt and removing/adjusting /boot/resolution.txt.

belliash, 2023-05-02:
One more question - is it possible to boot it from USB or it needs to be flashed on SD card?

KonstaT, 2023-05-02:
> One more question - is it possible to boot it from USB or it needs to be flashed on SD card?

You need to adjust /boot/config.txt to boot from USB. Please read the relevant FAQ section.

KonstaT, 2023-05-02:
Have you tested uncommenting `dtoverlay=rpi-backlight` in /boot/config.txt. That's what the official Raspberry 7" DSI touchscreen uses as backlight driver but IDK about these Waveshare ones.

belliash, 2023-05-02:
It uses Goodix driver which exposes some interface through /proc to control backlight. Nevertheless I will try with `dtoverlay=rpi-backlight` as well.

KonstaT, 2023-05-02:
Goodix driver is for the touchscreen. It doesn't have anything for backlight control.

In any case my Android lights HAL only supports `/sys/class/backlight/rpi_backlight/brightness` to control the display brightness using the brightness slider in Android UI. Otherwise you would need to echo the brightness value manually.

`dtoverlay=rpi-backlight` is commented out by default.

closed 2023-05-02 by KonstaT