lineage-rpi/android_kernel_brcm_rpi #34: Cannot disable dhcp or set static ip address.

msarvi, 2023-10-17:
### Describe the bug

I want to set **static IP address for ethernet**. i use `ifconfig` command using ADB and change the IP address but after reconnecting from ethernet or turn off/on android device the IP Changes.
I think this is cause by **DHCP** . I cannot a file for disable DHCP in android device. 

android device: rpi-4b
os: Android 12 Lineage OS.

### Steps to reproduce the behaviour

DHCP
Ethernet.
Static IP 

### Device (s)

Raspberry Pi 4 Mod. B

### System

firmware version: 
version 2a652e2a1ddcf0f358bc6278802ff7d8f1397c2e (clean) (release) (start_x)
OS: Lineage OS 12.0
kernel: Linux localhost 5.10.169-v8-g24a9f8401a3d #1 SMP PREEMPT Tue Mar 7 17:33:56 UTC 2023 aarch64

KonstaT, 2023-10-19:
Android 12 build are outdated and have been discontinued over six months. Please use the latest Android 13/14 builds instead.

I would say this is not generally supported in Android. Android TV has some ethernet options in the settings (dhcp/static ip). There isn't any in regular tablet UI builds.

How are you setting this on command line? Maybe adding a script to /vendor/etc/init.d/ or using some Tasker-like application could be used as a workaround?

msarvi, 2023-10-22:
> Android 12 build are outdated and have been discontinued over six months. Please use the latest Android 13/14 builds instead.
>
> I would say this is not generally supported in Android. Android TV has [...]

would you please help me to run script in init.d cause i have no idea, i add .sh file to /vendor/etc/init.d folder but i cannot find init.*.rc file for run the script in bash. 
unfortunately, There is no tutorial for this problem.
Thanks body.

msarvi, 2023-10-22:
i have found a file in /vendor/etc/init/hw/ with file name: `*rpi4-*.rc` and add script then each time after rebooting it set static ip for me. thanks for your help.

msarvi, 2023-10-25:
i updated to android 13 but cannot run scripts from init.d folder whic .rc file is for this path and where is .rc file

KonstaT, 2023-10-26:
You run script in /vendor/etc/init.d/ just fine. You need to make your script executable (755 permissions). There are already several scripts in that directory that you can use as a reference.

closed 2023-10-26 by KonstaT