lineage-rpi/android_kernel_brcm_rpi #35: PS4 Controller support

YoussF, 2024-01-11:
### Describe the bug

Hello,

I don't really know if i'm in the right place, I think i am. it seems that it's not possible to pair a bluetooth ps4 controller to the raspberry pi 4 under lieage-rpi (with Gapps for Android TV installed)

### Steps to reproduce the behaviour

Take a ps4 controller press the SHARE and the PS button at the same time (the controller ligh should blink in white)
Go to bluetooth setting, you should see the controller ready to be paired, click on it to begin the paring process, then nothing happen, where you should have a paring request to confirm (that never occurs). 

### Device (s)

Raspberry Pi 4 Mod. B

### System

255|rpi4:/ $ cat /etc/rpi-issue
cat: /etc/rpi-issue: No such file or directory
1|rpi4:/ $ vcgencmd version
VCHI initialization failed
255|rpi4:/ $ uanme -a
/system/bin/sh: uanme: inaccessible or not found
127|rpi4:/ $

KonstaT, 2024-01-15:
This is missing which LineageOS version? Android TV I also have to assume? Also you would need to attach the appropriate logs (`logcat` in this case).

GApps are not part of my releases. There's been some previous reports of issues with pairing bluetooth remotes on Android TV after installing GApps (e.g. https://github.com/opengapps/opengapps/issues/915#issuecomment-1236377741).

KonstaT, 2024-02-21:
Closing as there has been no activity in over a month.

closed 2024-02-21 by KonstaT