lineage-rpi/android_kernel_brcm_rpi #36: TV doesn't recognize CEC

ilya-fedin, 2024-01-22:
### Describe the bug

I have an old LG TV with Simplink support. I've captured some CEC sommunication between the rpi and the TV, hope it helps.

### Steps to reproduce the behaviour

I've just connected the rpi to the TV...

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
rpi4:/ # raspinfo
/system/bin/sh: raspinfo: inaccessible or not found
127|rpi4:/ # cat /etc/rpi-issue
cat: /etc/rpi-issue: No such file or directory
1|rpi4:/ # vcgencmd version
Jan 18 2023 12:28:54 
Copyright (c) 2012 Broadcom
version 658f02cc8edcb68a568273f05d2b6ceede181e15 (clean) (release) (start_x)
rpi4:/ # uname -a
Linux localhost 5.15.106-v8-g8d0475872f62 #1 SMP PREEMPT Tue May 2 12:00:51 UTC 2023 aarch64 Toybox
```

### Logs

```
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
       vendor-id: 57489 (0x0000e091)
Received from TV to Playback Device 1 (0 to 4): GIVE_DEVICE_VENDOR_ID (0x8c)
Transmitted by Playback Device 1 to all (4 to 15): DEVICE_VENDOR_ID (0x87):
       vendor-id: 3075 (0x00000c03)
Received from TV to Playback Device 1 (0 to 4): GIVE_DEVICE_POWER_STATUS (0x8f)
Transmitted by Playback Device 1 to TV (4 to 0): REPORT_POWER_STATUS (0x90):
       pwr-state: on (0x00)
Received from TV to Playback Device 1 (0 to 4): GIVE_PHYSICAL_ADDR (0x83)
Transmitted by Playback Device 1 to all (4 to 15): REPORT_PHYSICAL_ADDR (0x84):
       phys-addr: 3.0.0.0
       prim-devtype: playback (0x04)
Received from TV to Playback Device 1 (0 to 4): VENDOR_COMMAND (0x89):
       vendor-specific-data: 0x04 0x00
Transmitted by Playback Device 1 to TV (4 to 0): FEATURE_ABORT (0x00):
       abort-msg: 137 (0x89, VENDOR_COMMAND)
       reason: refused (0x04)
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
       vendor-id: 57489 (0x0000e091)
Received from TV to Playback Device 1 (0 to 4): GIVE_DECK_STATUS (0x1a):
       status-req: on (0x01)
Transmitted by Playback Device 1 to TV (4 to 0): FEATURE_ABORT (0x00):
       abort-msg: 26 (0x1a, GIVE_DECK_STATUS)
       reason: unrecognized-op (0x00)
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
       vendor-id: 57489 (0x0000e091)
Received from TV to Playback Device 1 (0 to 4): VENDOR_COMMAND (0x89):
       vendor-specific-data: 0x01
Transmitted by Playback Device 1 to TV (4 to 0): FEATURE_ABORT (0x00):
       abort-msg: 137 (0x89, VENDOR_COMMAND)
       reason: refused (0x04)
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
       vendor-id: 57489 (0x0000e091)
```

KonstaT, 2024-01-22:
Check your TV's settings. Does your TV have CEC menu where to select the device? You can also try uplugging other CEC devices on your TV or using the TV's primary HDMI port for the Pi. Also make sure that you're using the same HDMI port on the Pi that is set for HDMI-CEC in Raspberry Pi settings (HDMI0 by default).

ilya-fedin, 2024-01-22:
Yes, I think I use HDMI0 (the one right after the USB-C power plug) and I don't have any other CEC devices. The TV has CEC menu, it looks like this:
[image]
If i disable simplink (cec) and re-enable it, it shows load indicator for a couple of minutes, I see the connection attempts in the cec monitor like the ones I provided during that time and after.

It feels to me that the problem is either here
```
Received from TV to Playback Device 1 (0 to 4): VENDOR_COMMAND (0x89):
       vendor-specific-data: 0x04 0x00
Transmitted by Playback Device 1 to TV (4 to 0): FEATURE_ABORT (0x00):
       abort-msg: 137 (0x89, VENDOR_COMMAND)
       reason: refused (0x04)
```
Or there
```
Received from TV to Playback Device 1 (0 to 4): GIVE_DECK_STATUS (0x1a):
       status-req: on (0x01)
Transmitted by Playback Device 1 to TV (4 to 0): FEATURE_ABORT (0x00):
       abort-msg: 26 (0x1a, GIVE_DECK_STATUS)
       reason: unrecognized-op (0x00)
```

KonstaT, 2024-01-22:
Where are these logs even from? `cec-ctl`?

Yeah, I don't know about your TV. On mine I need to select the specific device (raspberrypi) from the CEC menu for it work.

This is the HDMI-CEC HAL used on my LineageOS 20 releases. You're welcome to contribute a fix.
https://github.com/raspberry-vanilla/android_device_brcm_rpi4/tree/android-13.0/cec

ilya-fedin, 2024-01-22:
> Where are these logs even from? `cec-ctl`?

Yes

> Yeah, I don't know about your TV. On mine I need to select the specific device (raspberrypi) from the CEC menu for it work.

I guess it should work automatically once the device is detected... But apparently it's not detected because rpi answers with FEATURE_ABORT to GIVE_DECK_STATUS. Apparently it's an essential info for this TV...

> You're welcome to contribute a fix.

Even if I'd manage to write the fix, I won't be able to build entire system with my hardware I guess...

ilya-fedin, 2024-01-22:
FWIW, powering on the TV works if I push the power off/on button via the Google TV Android application. Although it first powers on with the last used channel and switches to the right HDMI port only if I push the power off/on button two more times (to off/on it again). I also noticed that in the state when it should perhaps sleep, it shows black screen instead of stopping sending the signal to TV>

ilya-fedin, 2024-01-22:
I found some videos on youtube with example of how this is supposed to work:
[image]
I.e. some of those items should be enabled, you click on them for the check to appear, exit the menu and you can use the arrow keys and etc.

ilya-fedin, 2024-01-22:
> How exactly? If someone e.g. wanted to compare this to something that works.

`cec-ctl -m`

> https://source.android.com/docs/setup/start/requirements#hardware-requirements

I have a Ryzen 5 3500U / 6 GB RAM/ 250 GB SSD laptop. Looks like it's not fitting any requirements.

> You would only need to compile one file. `make hdmi_cec.rpi` produces '/vendor/lib(64)/hw/hdmi_cec.rpi.so' you can replace on existing images.

That's nice but I don't see actual message handling in the file. Is it in HAL at all? Maybe it's in Android itself?

> In any case, I don't have hardware to reproduce the issue so there's not much I can do to fix it either.

Can you make it answer something reasonable to GIVE_DECK_STATUS? I'd be happy to test. Pages 33-34 from https://web.archive.org/web/20171009194844/http://www.microprocessor.org/HDMISpecification13a.pdf (found in Google) seem to tell that it has to answer something like "Stop".

KonstaT, 2024-01-23:
You're also welcome to dig into the Android source (https://cs.android.com/ helps). Rest of the CEC HAL implementation is at https://android.googlesource.com/platform/hardware/interfaces/+/refs/tags/android-13.0.0_r75/tv/cec/1.0/ and the header at https://android.googlesource.com/platform/hardware/libhardware/+/refs/tags/android-13.0.0_r75/include/hardware/hdmi_cec.h.

Sorry, it makes no sense in time/effort wise stabbing this in the dark without the ability to test oneself.

ilya-fedin, 2024-01-24:
I found it: https://cs.android.com/android/platform/superproject/main/+/main:frameworks/base/services/core/java/com/android/server/hdmi/HdmiCecLocalDevice.java;l=296?ss=android%2Fplatform%2Fsuperproject%2Fmain&start=61

It sadly really doesn't have a GIVE_DECK_STATUS handler :(

Would it be a good idea to report to Android? If yes, what component it is? It has a lot of components in the list and I'm quite confused.

ilya-fedin, 2024-01-24:
Can you make a build (maybe of the respective framework only if possible?) with such a handler?
```
            case Constants.MESSAGE_GIVE_DECK_STATUS:
                byte[] param = new byte[] {
                        (byte) (0x89 & 0xFF)
                };
                mService.sendCecCommand(HdmiCecMessage.build(mDeviceInfo.getLogicalAddress(), message.getSource(), Constants.MESSAGE_DECK_STATUS, param));
                return Constants.HANDLED;

KonstaT, 2024-01-25:
> Can you make a build (maybe of the respective framework only if possible?) with such a handler?

> Sorry, it makes no sense in time/effort wise stabbing this in the dark without the ability to test oneself.

ilya-fedin, 2024-01-27:
Would a record from LibreELEC help?
```
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x00e091 (LG)
Transmitted by Recording Device 1 to all (1 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x00e091 (LG)
Received from TV to Recording Device 1 (0 to 1): FEATURE_ABORT (0x00):
        abort-msg: 0 (0x00, FEATURE_ABORT)
        reason: unrecognized-op (0x00)
Received from TV to Recording Device 1 (0 to 1): GIVE_DEVICE_VENDOR_ID (0x8c)
Transmitted by Recording Device 1 to all (1 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x00e091 (LG)
Received from TV to Recording Device 1 (0 to 1): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x01
Transmitted by Recording Device 1 to TV (1 to 0): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x02 0x05
Received from TV to Recording Device 1 (0 to 1): GIVE_DEVICE_POWER_STATUS (0x8f)
Transmitted by Recording Device 1 to TV (1 to 0): REPORT_POWER_STATUS (0x90):
        pwr-state: on (0x00)
Received from TV to Recording Device 1 (0 to 1): GIVE_PHYSICAL_ADDR (0x83)
Transmitted by Recording Device 1 to all (1 to 15): REPORT_PHYSICAL_ADDR (0x84):
        phys-addr: 2.0.0.0
        prim-devtype: record (0x01)
Received from TV to Recording Device 1 (0 to 1): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x04 0x00
Transmitted by Recording Device 1 to TV (1 to 0): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x05 0x01
Transmitted by Recording Device 1 to TV (1 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Transmitted by Recording Device 1 to TV (1 to 0): REPORT_POWER_STATUS (0x90):
        pwr-state: on (0x00)
Received from TV to Recording Device 1 (0 to 1): GIVE_DECK_STATUS (0x1a):
        status-req: on (0x01)
Transmitted by Recording Device 1 to TV (1 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Received from TV to Recording Device 1 (0 to 1): GIVE_DECK_STATUS (0x1a):
        status-req: once (0x03)
Transmitted by Recording Device 1 to TV (1 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x00e091 (LG)
Transmitted by Recording Device 1 to all (1 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x00e091 (LG)
Received from TV to Recording Device 1 (0 to 1): FEATURE_ABORT (0x00):
        abort-msg: 0 (0x00, FEATURE_ABORT)
        reason: unrecognized-op (0x00)
```

I've noticed multiple differences:
1. LibreELEC is a recording device rather than playback device
2. LibreELEC provides a valid answer to the vendor commands
3. LibreELEC provides a valid answer to the GIVE_DECK_STATUS with the value 0x20 what seem to conform to "Play Reverse" according to the page 87 of https://web.archive.org/web/20171009194844/http://www.microprocessor.org/HDMISpecification13a.pdf

This results in the RPI being detected as a HDD Recorder by the TV and remote control just works.

ilya-fedin, 2024-01-27:
Switched LibreELEC to playback device mode:
```
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x00e091 (LG)
Transmitted by Playback Device 1 to all (4 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x001582 (Pulse-Eight)
Received from TV to Playback Device 1 (0 to 4): FEATURE_ABORT (0x00):
        abort-msg: 0 (0x00, FEATURE_ABORT)
        reason: unrecognized-op (0x00)
Received from TV to Playback Device 1 (0 to 4): GIVE_DEVICE_VENDOR_ID (0x8c)
Transmitted by Playback Device 1 to all (4 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x001582 (Pulse-Eight)
Received from TV to Playback Device 1 (0 to 4): GIVE_DEVICE_POWER_STATUS (0x8f)
Transmitted by Playback Device 1 to TV (4 to 0): REPORT_POWER_STATUS (0x90):
        pwr-state: on (0x00)
Received from TV to Playback Device 1 (0 to 4): GIVE_PHYSICAL_ADDR (0x83)
Transmitted by Playback Device 1 to all (4 to 15): REPORT_PHYSICAL_ADDR (0x84):
        phys-addr: 2.0.0.0
        prim-devtype: playback (0x04)
Received from TV to Playback Device 1 (0 to 4): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x04 0x00
Transmitted by Playback Device 1 to TV (4 to 0): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x05 0x01
Transmitted by Playback Device 1 to TV (4 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Transmitted by Playback Device 1 to TV (4 to 0): REPORT_POWER_STATUS (0x90):
        pwr-state: on (0x00)
Received from TV to Playback Device 1 (0 to 4): GIVE_DECK_STATUS (0x1a):
        status-req: on (0x01)
Transmitted by Playback Device 1 to TV (4 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Received from TV to all (0 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x00e091 (LG)
Transmitted by Playback Device 1 to all (4 to 15): DEVICE_VENDOR_ID (0x87):
        vendor-id: 0x001582 (Pulse-Eight)
Received from TV to Playback Device 1 (0 to 4): FEATURE_ABORT (0x00):
        abort-msg: 0 (0x00, FEATURE_ABORT)
        reason: unrecognized-op (0x00)
Received from TV to Playback Device 1 (0 to 4): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x01
Transmitted by Playback Device 1 to TV (4 to 0): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x02 0x05
Received from TV to Playback Device 1 (0 to 4): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x04 0x00
Transmitted by Playback Device 1 to TV (4 to 0): VENDOR_COMMAND (0x89):
        vendor-specific-data: 0x05 0x01
Received from TV to Playback Device 1 (0 to 4): GIVE_DECK_STATUS (0x1a):
        status-req: on (0x01)
Transmitted by Playback Device 1 to TV (4 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Received from TV to Playback Device 1 (0 to 4): GIVE_DECK_STATUS (0x1a):
        status-req: once (0x03)
Transmitted by Playback Device 1 to TV (4 to 0): REPORT_POWER_STATUS (0x90):
        pwr-state: on (0x00)
Received from TV to Playback Device 1 (0 to 4): GIVE_DEVICE_POWER_STATUS (0x8f)
Transmitted by Playback Device 1 to TV (4 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Received from TV to Playback Device 1 (0 to 4): GIVE_PHYSICAL_ADDR (0x83)
Transmitted by Playback Device 1 to TV (4 to 0): DECK_STATUS (0x1b):
        deck-info: 32 (0x20)
Transmitted by Playback Device 1 to TV (4 to 0): REPORT_POWER_STATUS (0x90):
        pwr-state: to-on (0x02)
Transmitted by Playback Device 1 to all (4 to 15): REPORT_PHYSICAL_ADDR (0x84):
        phys-addr: 2.0.0.0
        prim-devtype: playback (0x04)
```
The remote still works, LibreELEC is recognized as a disc now.

ilya-fedin, 2024-01-27:
The VENDOR_COMMAND communication looks pretty much like https://github.com/Pulse-Eight/libcec/blob/bf5a97d7673033ef6228c63109f6baf2bdbe1a0c/src/libcec/implementations/SLCommandHandler.cpp

First the TV sends `SL_COMMAND_INIT` (`0x01`), libcec answers with `SL_COMMAND_ACK_INIT SL_COMMAND_TYPE_HDDRECORDER` (`0x02 0x05`). Then the TV sends `SL_COMMAND_CONNECT_REQUEST` (`0x04 0x00`) and libcec answers with `SL_COMMAND_SET_DEVICE_MODE CEC_DEVICE_TYPE_RECORDING_DEVICE` (`0x05 0x01`)

I've noticed that the CEC connection is unstable when LibreELEC is in playback device mode, maybe that's why