lineage-rpi/android_kernel_brcm_rpi #37: feature request: add support for Raspberry Pi TV HAT

bam80, 2024-02-11:
### Describe the bug

Could we add support for Raspberry Pi TV HAT?
https://www.raspberrypi.com/products/raspberry-pi-tv-hat/

I suppose we would need [Tuner HAL](https://source.android.com/docs/devices/tv/tuner-framework#tuner-HAL) implementation for that, I also asked in the forum:
https://forums.raspberrypi.com/viewtopic.php?t=365478

### Steps to reproduce the behaviour

On Android (TV) on RPi, you can't make use of Raspberry Pi TV HAT.

### Device (s)

Raspberry Pi 4 Mod. B

### System

https://konstakang.com/devices/rpi4/LineageOS20-ATV/

KonstaT, 2024-02-21:
I don't have the hardware to develop/test so there's unfortunately nothing I can do.

You can look into Raspberry Vanilla project to implement additional hardware support.
https://github.com/raspberry-vanilla

closed 2024-02-21 by KonstaT

bam80, 2024-02-21:
> You can look into Raspberry Vanilla project to implement additional hardware support.
https://github.com/raspberry-vanilla

OK thanks, do you think I could ask for the support there?

I doubt I can do the actual work myself, but could provide remote access to the board and and assist with the testing, maybe.

KonstaT, 2024-02-21:
I doubt I'm going to give you any different answer there. :)

No, there's nothing any developer can do without having the hardware themselves.